realthunder/FreeCAD_assembly3 #914: Rendering issue with 0.11 causes all faces to be transparent

0x0203, 2021-11-10:
When running the latest 0.11 stable branch, the rendering of geometry results in all faces being somewhat transparent, (see picture) regardless of draw style or transparency levels assigned to objects, bodies, etc... This happens with default settings and a clean install. It doesn't happen with the last linkstage release 0.10.2, nor does it happen with stock freecad 0.19. I'm running this under windows 7 on slightly older hardware, so I suspect that may be part of the issue. Does the new renderer rely on features only found in the latest graphics APIs?

[image]

Expected draw style:
[image]

realthunder, 2021-11-11:
Can you please post your FreeCAD About information here?

Also please post your application setting here. `Tools -> Edit parameters -> Export`.

0x0203, 2021-11-11:
About info: 
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +3962 (Git)
Build type: Release
Branch: LinkStage3
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US

Application settings are attached. It should be out-of-the-box stock, with no changes as I deleted the FreeCad directory from AppData/Roaming before testing. 
[freecad_render_issue.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7523387/freecad_render_issue.zip)
)

0x0203, 2021-11-17:
Coincidentally, my graphics card just died on me and I was forced to get a new one. Somewhere between the new card and new drivers, I no longer seem to have an issue. I'll close this assuming the problem was on my end.

closed 2021-11-17 by 0x0203