realthunder/FreeCAD_assembly3 #916: [FR] guide curve blend Surface

hamza00789, 2021-11-11:
Free Cad blend blend Surface

https://user-images.githubusercontent.com/92737149/141329516-f9e5f468-8362-4932-b29a-f089d4976b71.mp4

guide curve blend Surface 

https://user-images.githubusercontent.com/92737149/141329633-cc8a3080-4b23-4e75-aa31-7ce3a6035680.mp4

Rabbit-sk, 2021-11-11:
[Curves Workbench – FreeCAD Documentation](https://wiki.freecadweb.org/Curves_Workbench)

[FreeCAD Curves and Surfaces WorkBench](https://github.com/tomate44/CurvesWB#freecad-curves-and-surfaces-workbench)

[Issues · tomate44/CurvesWB](https://github.com/tomate44/CurvesWB/issues)

Rabbit-sk, 2021-11-11:
It seems that your model is not solid, but only consists of surfaces. It is possible to model this in PartDesign, but it works better with solids. 

If this part is to be made of sheet metal, there is a workbench for this type of item: [https://wiki.freecadweb.org/SheetMetal_Workbench](https://wiki.freecadweb.org/SheetMetal_Workbench)

Rabbit-sk, 2021-11-11:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Rabbit-sk, 2021-11-11:
Was it helpful?

hamza00789, 2021-11-11:
> 
> 
> Was it helpful?

Sorry Couldn't talk due to busy schedule. I gave a simple example so that I can understand what I want. In this example this work can be done in part design and partworkbanch as you did. And I don't have to do anything like sheet metal. I have to work on the surface. My work is being stopped because of this. I try to give a good example of this. 

https://user-images.githubusercontent.com/92737149/141363148-3130eb7d-c048-4d87-9d56-8a68125fcd20.mp4

mwlandolsi, 2021-11-15:
Sorry to  comment out of the subject, but I was interested by the way that the annotations are displayed
Could you please tell me @hamza00789  which workbench you used to create them.
[image]

hamza00789, 2021-11-15:
> 
> 
> Sorry to comment out of the subject, but I was interested by the way that the annotations are displayed Could you please tell me @hamza00789 which command and workbench you used to create them. [image]

Why do you want to know this? It will not solve my problem

closed 2021-12-06 by hamza00789

hamza00789, 2022-01-20:
> Sorry to comment out of the subject, but I was interested by the way that the annotations are displayed Could you please tell me @hamza00789 which command and workbench you used to create them. [image]

https://youtu.be/PwUYf0B5rSE?t=201