realthunder/FreeCAD_assembly3 #919: window for active clipping plane cant be reopened after closing

HoWilgh, 2021-11-12:
FreeCAD_assembly3 enables many new settings like (Overlay or dock-window) for the sub-windows like the clipping plane.

The problem arises as the window for the clipping plane now can be closed using the X of 'Close dock window' without disabling an active clipping-plane. Therefore the clipping plane (not the window) stays active and the window for the clipping plane cant be reopened (View->Clipping plane). 
The only way to get clipping plane back is to restart FreeCAD!
This is only an issue of FreeCAD_assembly3 and not of FreeCAD because there you have to press the close button which also disables the active clipping plane.

Usually this arises when working with different files and windows opened where one needs to use clipping plane for different files .... so you want to close the window to use it for a different file/window.

realthunder, 2021-11-12:
Can you please record a screencast to show the problem? I can't seem to reproduce it. For me, I can reopen the clip control dock after close it with the X button. BTW, you don't need to close the panel when switching to a new document. The panel shows independent settings for each document. To de-activate the clip plane, click the `Close` button at the bottom of the clip control dock. Or, simply press `ESC` when you are focusing on the control dock.

HoWilgh, 2021-11-12:
You need to let the clip stay active and close the window with the X.
[image]

realthunder, 2021-11-12:
Oh, you are using the `Stable` release. If you use clip plane often, you may want to try my `Daily` release. There is major enhancement in clip function there. It hasn't been merged to my `Stable` release. With the `Daily` release, please first set `Preference -> Display -> Render cache` to `Experimental`.

HoWilgh, 2021-11-13:
Sorry for not explicitly mentioning that I use Stable.
I know there are significant enhancements in Daily but I use mostly Stable to circumvent problems when new features are introduced.

mwlandolsi, 2021-11-18:
Hello 
Even on daily branch, closing the window of the clipping planes by "X" button on the top does not close the clipping view. I thin  that the cliping view should be closed also when clicking on closing "X" at the top.
https://user-images.githubusercontent.com/76944804/142489216-e5f01af7-a7c7-401c-94bc-ffe7fc5fa9cd.mp4

maxwxyz, 2023-01-30:
@realthunder is there a timepline when your awesome clipping plane features will be merged into main FreeCAD? Is it done with the current topnaming branch merge?

realthunder, 2023-01-30:
Unfortunately, there is no timeline yet. I do plan to submit it eventually. It is going to be a very big PR. It's not just the clipping view, but an entire new renderer.

@mwlandolsi Sorry, missed your post for so long. Hope you've already realized that hitting the 'Close' button can close the clipping view. The reason for the behavior of the 'X' button is that sometime people want to hide the clipping panel to enlarge the view without deactivate the clipping view.

maxwxyz, 2023-01-30:
@realthunder ok but would be epic! Is there anything I could help you with? I'm no programmer but maybe in testing or sth. else?