realthunder/FreeCAD_assembly3 #922: Export issue in FreeCad

hamza00789, 2021-11-13:
Free Cad export command does not work properly.

oficinerobotica, 2021-11-13:
@hamza00789 Sorry but I couldn't help my self. I understand that language can be a barrier, you might not be a native English speaker and I  also struggle with that allot but sincerely opening issues like this isn't helpful.  You need to provide a bit more info and a example file, image or anything more than that for anyone to even remotely comprehend your issue. 

FreeCAD can export in allot of formats. What format have you tried? What program have you used to open your export? Can you provide an example file?

 What version of FC are you using?  That is also a very important info needed. 
You might have bumped in to a real issue but no one can help with that little info at hand.
Best regards.

hamza00789, 2021-11-13:
> @hamza00789 Sorry but I couldn't help my self. I understand that language can be a barrier, you might not be a native English speaker and I also struggle with that allot but sincerely opening issues [...]

FreeCAD version [FreeCAD-asm3-Daily-Win64-Py3-Qt5-20211015]  
File 
[FreeCad File.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7532352/FreeCad.File.zip)   
You can see I am export free cad file [file type .stp] 

https://user-images.githubusercontent.com/92737149/141651654-4d0c978f-8ddf-4055-ac87-97c7393e312e.mp4            

I am opening export file in Catia V5. You can check your result  

Uploading 2.mp4…
 
I am opening export file in delcam powershape. You can check your result

https://user-images.githubusercontent.com/92737149/141651788-4206417e-2794-4ef4-b041-2933023cf013.mp4    
 
The file that is exported from FreeCad when the file is opened in other software then some parts of it are missing

herrgerd, 2021-11-13:
You already opened an issue for that here: https://github.com/realthunder/FreeCAD_assembly3/issues/915
Please close this one or the other?!

hamza00789, 2021-11-14:
> 
> 
> You already opened an issue for that here: #915 Please close this one or the other?!

I just found out why some parts of the file are missing. 
But a new problem has emerged.           

Is a solution possible?

https://user-images.githubusercontent.com/92737149/141673793-fd94b868-7c9e-451a-9a01-36d2ad3667ae.mp4

hamza00789, 2021-11-15:
> 
> 
> > You already opened an issue for that here: #915 Please close this one or the other?!
> 
> I just found out why some parts of the file are missing. But a new problem has emerged.
> 
> Is a solution possible?
> 123.mp4
> 
> My Free Cad Version OS: Windows 10 Version 2009 Word size of OS: 64-bit Word size of FreeCAD: 64-bit Version: 2021.1015.24301 +4280 (Git) Build type: Release Branch: LinkDaily Hash: 556c87868ea46796242156e9f73eae98259794b7 Python version: 3.8.6+ Qt version: 5.15.2 Coin version: 4.0.1 OCC version: 7.5.0 Locale: English/United States (en_US)

FreeCAD uses OCC for import/export of step, there have been regressions in OCC. My recollection is that one need OCC 7.5.2 before things were fixed. In the Daily 10.15 releases notes it says step issues fixed, so is this really 7.5.0 or are the problems not all fixed?

realthunder, 2021-11-15:
OCC indeed got quite a few STEP related regressions in 7.5. I'll try upgrade the OCC version for Windows in next release.

hamza00789, 2021-11-15:
> OCC indeed got quite a few STEP related regressions in 7.5. I'll try upgrade the OCC version for Windows in next release.
 
thanks

hamza00789, 2021-11-15:
> 
> 
> > OCC indeed got quite a few STEP related regressions in 7.5. I'll try upgrade the OCC version for Windows in next release.
> 
> thanks

[1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7541887/1.zip)

hamza00789, 2021-11-15:
> 
> 
> > > OCC indeed got quite a few STEP related regressions in 7.5. I'll try upgrade the OCC version for Windows in next release.
> > 
> > 
> > thanks
> 
> [1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7541887/1.zip)
> 
> I did the ridge geometry of the file. But the STEP file still turned out to be bad
> 
> As you can see, there is no error now
> 
> [image] 

I exported this file which is shown in the picture and opened it in other software you can see the result 

[image]

closed 2021-12-21 by hamza00789