realthunder/FreeCAD_assembly3 #923: [FR] hide/Show

hamza00789, 2021-11-15:
Some frequently used specific items. in one click hide all in one click show all. for example sketches surfaces solids and curves etc. This makes it much easier to work when the work is long enough For example, if I need to edit a sketch, I can see where it is in the tree. It can take a long time to find it in a tree.

realthunder, 2021-11-15:
There is a search function in tree view. Just focus on the tree view, and press `Ctrl + F`. Type the object name or its label. There will be auto completion to help you along the way. You can also use object path to narrow the search, like `Fusion.Cut.Box`.

hamza00789, 2021-11-15:
> 
> 
> There is a search function in tree view. Just focus on the tree view, and press `Ctrl + F`. Type the object name or its label. There will be auto completion to help you along the way. You can also use object path to narrow the search, like `Fusion.Cut.Box`. 
[You can also use object path to narrow the search, like `Fusion.Cut.Box`] I do not understand this part of you

closed 2021-11-15 by hamza00789