realthunder/FreeCAD_assembly3 #925: The tip of the body is not updated correctly

Rabbit-sk, 2021-11-16:
Dear RealThunder,

I noticed that after change of a pad back to additive type, the tip is not removed from previous solid feature.

https://user-images.githubusercontent.com/74265320/141970796-cb4c9a25-6887-405b-bc64-01960fd91fa3.mp4

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Slovak/Slovakia (sk_SK)

Rabbit-sk, 2022-01-10:
Hi @realthunder !
Is there anything new about this?

Rabbit-sk, 2022-01-12:
Hi @realthunder !

It is the same in the new version 2022.01.11.

luzpaz, 2022-04-08:
@Rabbit-sk still reproducible on latest LinkDaily?

Rabbit-sk, 2022-04-09:
It behaves the same way. Now I have noticed that the color of the tags changes correctly (after changing the next solid feature from `New Shape` to `Additive`, the color of tags are the same) and so also the tip of the body changes correctly. I'm sorry I incorrectly described the problem at the beginning. 

In my opinion, the color of the icon tag is not noticeable enough because there is little space for its color. The size, shape, and edge lines of the icon, including the circle and quite big rectangle for the green tip of the body, and the shading of the icon leave little room for the recognition of the color of the tag icon. Maybe the tag icon could be redesigned. But that's a side issue.

The updated description of the problem (which do not have a negative impact on functionality) is as follows:
The tag icon next to the previous solid feature does not disappear after changing the next solid feature from `New Shape` to `Additive`.

So after changing a solid feature from `New Shape` to `Additive` instead of this: 
[image]

the outcome would be like this:
[image]

But this proposal is probably not correct. This is probably the behavior of tag icons if from body becomes multibody. Now I have noticed that tag icons are added to all previous features if another solid feature is set to New shape:
[image]
[image]

Removing tag icons for previous features, for example after changing Pad004 to Additive, would be appropriate only if there are no other features with a different tag color in the given body. This is therefore a marginal case of working with multibody, which could become singlebody.

The final proposal could be:
**Remove tag icons next to the previous solid features after changing the last solid feature from `New Shape` to `Additive`, if there is no features with different tag color in a given body.**