realthunder/FreeCAD_assembly3 #926: Crash after selecting objects after deleting object after it was set to show on top

Rabbit-sk, 2021-11-16:
Dear RealThunder,

if object is set to show on top, and after then it is deleted and after then mouse is used to select object in 3D view or tree view, FreeCAD crashes.

https://user-images.githubusercontent.com/74265320/141973734-7533463d-6e61-48cc-a7e5-1e76ddd75393.mp4

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Slovak/Slovakia (sk_SK)

Rabbit-sk, 2022-01-09:
Hi @realthunder !
Is there anything new about this?

realthunder, 2022-01-09:
There are quite a few fixes regarding show-on-top option starting from https://github.com/realthunder/FreeCAD/commit/c8fa31a66255ad516eb714078eb04d241db8dccc. The problem reported in this issue is most likely fixed.

PS. thanks do your reminder, I tried to repeat the steps shown here, and fixed a newly introduced bug in recent commits.

closed 2022-01-09 by realthunder

Rabbit-sk, 2022-01-10:
> PS. thanks do your reminder, I tried to repeat the steps shown here, and fixed a newly introduced bug in recent commits.

😎