realthunder/FreeCAD_assembly3 #928: Assembly 3 writing invalid XML to Document.xml

hallgchris, 2021-11-17:
Hi there! I've found a bug which I believe to be caused by Assembly 3 - I'm pretty new though, and don't have the knowledge to verify this.

Basically, the issue I experienced amounts to the `Document.xml` of one of my `FCStd`s contained a `<` character in a string, rather than being escaped as `&lt;` as I think it should be. This caused the assembly/model to fail to load. Manually replacing the character with this escape sequence made loading successful, but upon saving and reopening the file was invalid again.

The invalid character was on a line that looked like this:
```xml
        <Property name="LinkedObject" type="App::PropertyXLink" group=" Link" attr="0" ro="0" hide="0" doc="" status="2097412">
          <XLink file="" stamp="" name="Link001" sub="1.$BBMidPlane@ClampRing001<1>."/>
        </Property>
```

Corresponding to this in my assembly.

[image]

As far as I can tell, the `<1>` appears because the link is to the copy of `ClampRing001` with index `1`.

[image]

If it's useful for reproducing, here's my actual FreeCCAD assembly. The issue is in `BBGantry.FCStd`, specifically line 10696 of `Document.xml` inside it.

[Bike jig bug.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7554144/Bike.jig.bug.zip)

<details>
<summary>FreeCAD/system details</summary>

OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +3962 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/Australia (en_AU)

</details>

I hope this is useful! Let me know if I can help further at all, and apologies for the big info dump. And don't assume anything I've written above is true, I've just made a lot of best guesses!

By the way, in case anyone else encounters this and is looking for a fix - Temporarily fix the xml (look for the line number in the report view, open the FCStd in an archive viewer, replace the bad character at that line in `Document.xml`) so that it can open in FreeCAD correctly. Locate whatever link is causing the issue, and rename the link so that it no longer contains a `<`.

realthunder, 2021-11-19:
Thanks for the detailed report. It is fixed in the above commit.

closed 2021-11-20 by realthunder

j-hap, 2021-11-21:
This seems closely related to #852 Any idea when or if this will be pulled into the upstream FreeCAD project (not "just" your fork)?

realthunder, 2021-11-21:
Sure. That's the reason I added `upstream` label, which indicates my intention for a PR to upstream. I'll do that soon.

hallgchris, 2021-11-22:
Thanks for the quick fix!