realthunder/FreeCAD_assembly3 #929: [FR] In Part Design Features

hamza00789, 2021-11-17:
Part design workbench in Close surface thickness surface There should be such commands

mwlandolsi, 2021-11-18:
Hello Hamza,
The video liked above is for catia and is related to an open surface and thsi is available also in FreeCAD.
Hope that this screen record could help you
https://youtu.be/WPKJYfQseH0

hamza00789, 2021-11-18:
> 
> 
> Hello Hamza, The video liked above is for catia and is related to an open surface and thsi is available also in FreeCAD. Hope that this screen record could help you https://youtu.be/WPKJYfQseH0 

What you are saying is very different from what I want. I know it converts from surface to solid in freecad. This command exists in another workbench which I want but does not work properly there so I expressed my desire in part design workbench

hamza00789, 2022-02-12:
I don't want that kind of thickness. This is the thickness of the surface that acts as a pad. 

[image] 

I want extended thickness. Also, extend the faces with the thickness of the upper face. The second is that this feature should be in the part design. 

For example 

[image]