realthunder/FreeCAD_assembly3 #934: disable sync/freeze of spreadsheet-link

HoWilgh, 2021-11-19:
I created a spreadsheet in my main-assembly-file and inserted links of this into the files of my sub-assemblies.
Now I cant find any possibility to freeze the links of the original spreadsheet to avoid having 'cyclic dependencies'.

realthunder, 2021-11-20:
Can you please provide a file showing the problem?

HoWilgh, 2021-11-21:
in the .zip you find two files tmp1.fcstd and tmp2.fcstd.
tmp1 is the main assembly and tmp2 is a subassembly linked into tmp1.
tmp1 contains a spreadsheet which is 'linked' into tmp2 so a variable can be used in a sketch in a body in tmp2.
A circular dependency is created  if one uses a shapebinder (Binder(Sketch)) in the linked tmp2 in tmp1.  
By freezing the linked spreedsheet in tmp2 this problem would be circumvented.

[tmp_assembly_with_spreadsheet.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7576858/tmp_assembly_with_spreadsheet.zip)

realthunder, 2021-11-22:
I am not sure what you mean by 'freezing the linked spreadsheet'. If tmp1 is the main assembly, then the suggested workflow is to only create Link in tmp1 that points to other sub-assembly file, not the other way round, which will create a dependency loop. Any reason for doing that?

HoWilgh, 2021-11-22:
I want to have all parameters for a 'main' assembly in the main file. Then I distribute them to the child-assemblies which makes the children dependent on the mainAs (one direction).
If one child is has e.g. a shapebinder from the main-assembly than we get a dependency loop, or? This way all assemblies work in one or the other way.
Therefore, i would like to freeze the link to the spreadsheet as it can be done with shapebinders.
Hope this helps.

realthunder, 2021-11-22:
I checked your files again. I can see now that you are trying to use the master sketch design pattern. In order for it to work, the master sketch must be in its own file and be added to the main assembly as another sub-assembly. You can fix the master sketch assembly using a `Lock` constraint. You can then to put your master spreadsheet into this `master sketch sub-assembly`, and reference it in other sub-assembly files without dependency loop.

HoWilgh, 2021-12-25:
First, it only works the way you suggested if I do not use the values of the spreadsheets together with the 'master sketch sub-assembly' in the sub-assemblies.
Second, this is not a intuitive procedure (at least for me).

[example_with_master-sketch_and_spreadsheet.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7776056/example_with_master-sketch_and_spreadsheet.zip)
I modified my model from above to show the result. 
I now have 'As tmp3 Mastersketch' linked into the main assembly. In main i use the mastersketch in another linked assembly (As tmp2, also using the data from spreadsheet) as well as in an internal assembly (Assembly a sub-assembly). I want to create holes in both assemblies which are linked to the main-sketch which is using the Spreadsheet. 
Finally, in main in 'As tmp2, also using the data from spreadsheet' I had to detach the facebinder to the mastersketch to agein avoid a cyclic dependency.

I hope all this is not to obscure....

BTW: The 'linkage' of spreadsheets is not shown in the dependency-graph, what makes it hard do grasp.