realthunder/FreeCAD_assembly3 #938: External geometry issue

hamza00789, 2021-11-22:
External geometry Geometry is not complete. Or the geometry of the other side

https://user-images.githubusercontent.com/92737149/142896965-5241c366-7de4-474e-866b-1b163d31780c.mp4

[image]
[image]
[image]
[image]

bambukouk, 2021-11-23:
see also #933

bambukouk, 2021-11-23:
my head is spinning :-)
is it closed or re-opened ....

hamza00789, 2021-11-23:
> my head is spinning :-) is it closed or re-opened ....

Sorry. It has been opened.

bambukouk, 2021-11-23:
Haven't got a clue :-(
@realthunder never replied so I don't even know whether he is doing anything about this one?
Hopefully he will surprise us (positively) as he often does :+1:

realthunder, 2021-11-24:
@bambukouk @hamza00789 Can you please post the files that have the problem? I've recently merged with upstream 0.20. So maybe the problem is solved. I'll check.

bambukouk, 2021-11-24:
@realthunder 
Thank you :+1: 
I have posted the file (plus additional explanations) in the original issue https://github.com/realthunder/FreeCAD_assembly3/issues/933#issuecomment-977643515 just in case it is separate from @hamza00789  issue?

hamza00789, 2021-11-24:
> @bambukouk @hamza00789 Can you please post the files that have the problem? I've recently merged with upstream 0.20. So maybe the problem is solved. I'll check.

[Bolt 2.1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7597051/Bolt.2.1.zip)

closed 2021-11-30 by realthunder