realthunder/FreeCAD_assembly3 #941: Questions related to attachment constraint

GinTonic99, 2021-11-24:
Good day,

OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: German/Austria (de_AT)

in an effort to use the attachment constraint as much as possible, I started investigating the beahaviour of the attachment constraint used with edges. Attached is a simple model comprising a grey pad to be connected to a blue bar with triangular cross section.

 I move the bar closed to the desired position with the correct orientation using the transformation tool and select the two edges to be connected.
[image]

After applying the attachment constraint this is the result.
[image]

As shown the bar has been rotated by 90°. In order to correct this I try to apply a 90° rotation around the z-axis for either one of the concerned elements. However this does not bring any change. Rotations around the x or y axis are observed but around the z-axis for either of the two elements not. Same when trying the offset element tool. However the flip element tool works but does not lead to the intended result in this case.

Question 1: How can I apply such a rotation around the z-axis for the elements?

During playing around I noticed that any input for the offset in of the two representations of an element in the tree (under the respective constraint and under Elements) is not reflected in the other representation. E.g. when I enter an offset of 10 mm in z direction in the element under the constraint, the offset of the element under Elements is still zero and when I enter another offset in the element under Elements this offset is applied in addition. My understanding has been that these two representations should be identical, like when I change the label in one it is reflected in the other representation.

Question 2: Is this the intended behaviour?

Further experiments show, that a rotation applied directly in the linked part would lead to the desired result.
[image]
I noticed however that it is not possible to enter the values directly, they have to entered in the expression field in order to be observed.

Question 3: is this the intended behaviour?

The enclosed file reflects that status after application of the rotation in the linked part.

Thank you for reading.

best regards
[Attachment.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7594553/Attachment.zip)

bambukouk, 2021-11-25:
You are complicating things :-)
I took your file, renamed things (to make it easier to follow things).
I also changed the triangle shape to easier follow which way round things are.
You don't need to rotate, transform or introduce placement formulas (yes, I have deleted them as well)
Then as you did, I "locked" and "attached" (without even getting things close together) - worked first time, and...
if I didn't as it sometimes happens - use "flip element" "flip part" "offset part" to get things orientated the way you desire
[Attachment(bambuko).zip](https://github.com/realthunder/FreeCAD_assembly3/files/7603899/Attachment.bambuko.zip)
[image]

Same model with element flipped:
[image]

or you could combine above with using scale=-1 (mirror)
[image]
[image]

I love assembly 3 :+1: :1st_place_medal:

GinTonic99, 2021-11-25:
Hi Bambukouk,

thank you for your effort.

What I was missing was exctly this:

> or you could combine above with using scale=-1 (mirror) !

thanks again
gintonic99

closed 2021-11-25 by GinTonic99

bambukouk, 2021-11-25:
You are welcome :+1:  @GinTonic99