realthunder/FreeCAD_assembly3 #946: Thickness problem

hamza00789, 2021-11-26:
The freecad thickness command does not work properly.

bennvan, 2022-01-12:
The inner thickness generates fine you check the intersection box when creating the feature. Robustness with the thickness feature seems more like an issue with occ.

hamza00789, 2022-01-12:
> 

It works with the intersection, but it changes something 

[image]

hamza00789, 2022-01-12:
> The inner thickness generates fine you check the intersection box when creating the feature. Robustness with the thickness feature seems more like an issue with occ.        

It works with the intersection, but it changes something  

[image]

bennvan, 2022-01-13:
> It works with the intersection, but it changes something

That's because the join type is set to arc. You said in your original post it works well for outer thickness but in your video edge radius are added in the exact same way?
Selecting intersection from the join type dropdown prevents this behaviour but for this specific geometry it locks up freecad for me. Again, this seems really an issue with OCC.
Unfortunately for now it looks like you'll have to model your geometry another way.

@luzpaz Maybe you can label this as OCC or close the issue?

hamza00789, 2022-01-13:
> That's because the join type is set to arc. You said in your original post it works well for outer thickness but in your video edge radius are added in the exact same way? Selecting intersection [...]

What is meant by the OCC label? Change its title

luzpaz, 2022-01-14:
OCC is the CAD kernel FreeCAD uses. More info at https://wiki.freecad.org/OpenCASCADE