realthunder/FreeCAD_assembly3 #947: white space expected - fatal error

bambukouk, 2021-11-28:
Luckily for me I save everything in a new version before attempting anything...
so it is not a total disaster - just go back to previous version...
but, I am totally puzzled by this error.
help please, pretty please :+1: 

[image]
[image]

realthunder, 2021-11-28:
Looks like illegal characters in XML file. If so the problem has been fixed. Until the next release, please do not use the following  characters for element name. 

```
<
>
&
'
"
```

bambukouk, 2021-11-28:
Yes, indeed 
I have used " to indicate imperial sized nipple 
Thank you
and I thought it was another one of my mistakes...
BTW google search came up with one page about XML, but I couldn't make sense of it 🤪

closed 2021-11-28 by bambukouk