realthunder/FreeCAD_assembly3 #949: Sketch is not moving in part design wb

drom89, 2021-11-29:
Sketch is not moving in part design wb.

[image]
[vozík.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7621841/vozik.zip)

drom89, 2021-11-29:
When i deleted all of external geometries it started to work.

kevenwyld, 2021-11-30:
In the video you can see the red exclamation point on Groove002 which is causing a failure to recompute.  Unfortunately you can't see the failure other than this small red icon because you don't have the status bar or report view turned on.

I opened your file and selected a valid axis for the revolve in Groove002 and everything started to work as expected.

Also for anyone that might try to open the file, it's a winRAR file but the extension is ".zip".

drom89, 2021-11-30:
You are right. i didnt try to change the axis because it says Invalid shape name not invalid axis or something :).
Thanks

closed 2021-11-30 by drom89