realthunder/FreeCAD_assembly3 #952: "axial move part" - for locked and frozen assemblies

bambukouk, 2021-12-01:
I have an `"ass_C(top_level)"` with two parts `"subass_A(anchor)"` and `"subass_B"`.
Both `"subass_A(anchor)"` and `"subass_B"` are locked and frozen.

When I try to use `"axial move part"` (in assembly3) to move `"ass_C(top_level)"` it is rightly greyed out and impossible to use, but when I try the same on the two sub-assemblies the result depends on whether they are active or not.

When either `"subass_A(anchor)"` or `"subass_B"` are active, the `"axial move part"` command is greyed out (quite correctly indeed), but... to my surprise, if they are not active it is quite possible to move them using this command???

Should it be so? and why?
As far as I am concerned once assembly is locked and frozen it should not be possible to move it using assembly 3 `"axial move part"` command regardless whether they are active or not?

Remain puzzled :+1:
[ass_C(top_level)zip](https://github.com/realthunder/FreeCAD_assembly3/files/7633883/ass_C.top_level.zip)

realthunder, 2021-12-02:
You need to enable `Lock mover` option to gray out mover on fixed assembly.

[image]

PS. I notice that you are using binder to derive the model of one part from another part in the same assembly while frozen. This may leads to some confusion. As the `freeze` does not prevent parts from updating. It only prevent the assembly from updating its snapshot of its parts. So you may be able to see both the frozen snapshot and the update parts in some cases. The `Freeze` function is mostly intended to save document loading time for sub-assembly in external document, in which case, the parts of a frozen assembly in external document will not be loaded at all, so there will not be any confusion.

bambukouk, 2021-12-02:
Thank you :+1: 

> You need to enable `Lock mover` option to gray out mover on fixed assembly.

What do you mean by `fixed` assembly?
Do you mean locked? using` Locked constraint`?
Or do you mean `to fix assembly`by enabling `Lock mover`?

> PS. I notice that you are using binder to derive the model of one part from another part in the same assembly while frozen. 

Not quite,
I have used binders to derive one part from another before freezing the assembly afterwards.

Understand your concern about possible confusion, but I cannot create my models without the use of binders???
or at least, to me it defeats the purpose of CAD as used for creation of related parts?
Are you telling me that I shouldn't use `freezing` because I use binders in my designs?

>As the `freeze` does not prevent parts from updating. It only prevent the assembly from updating its snapshot of its parts.

Understand - this is logical, although in my experience when I tried to modify frozen sub-assemblies nothing was happening/updating until I temporarily un-frozen it...?

> The `Freeze` function is mostly intended to save document loading time for sub-assembly in external document

Thank you for the explanation.
Same like "Locked constraint" that doesn't lock but just `...prevent the assembly solver from changing the part's location....`, the Freeze function doesn't freeze just `"saves loading time"` :-)

I suppose for now (until the design is totally completed) I might as well not bother using `Freezing function`?

realthunder, 2021-12-03:
> What do you mean by `fixed` assembly?
> Do you mean locked? using` Locked constraint`?

I mean those under `Locked constraint`. Actually, a part without any constraints is also considered as `Fixed`. The `Lock constraint` is only necessary if the part is constrained against other parts. 

> Are you telling me that I shouldn't use `freezing` because I use binders in my designs?

Kind of. I suggest you separate the process of modeling and assembling. If you want to use binder to do `in context` modeling, then it is better to make an assembly that is dedicated for this purpose. You then create a separate assembly in another file to use the model as part for assembling. This will reduce the chance of accidently change your model while assembling.

> I suppose for now (until the design is totally completed) I might as well not bother using `Freezing function`?

The `freeze` function really only make sense if you use the part through external link. Imagine a sub-assembly with many low level hierarchies. Once your design of that component is done, you don't really need all those low level details. The `freeze` function lets you get rid of those when loading the component in the main assembly file.

bambukouk, 2021-12-03:
> ...a part without any constraints is also considered as `Fixed`. 
> The `Lock constraint` is only necessary if the part is constrained against other parts.

great explanation!
actually this is exactly what I am doing in my models
just that sometimes I felt "guilty" if there was no locked (or any other) constraint [image] 
for where assembly 3 was simply used as a container without it's parts being constrained against other parts.

> ...I suggest you separate the process of modeling and assembling. 

Understand, but... (and it is a big but for me) all my parts are modeled `in context` to other parts.
It is not as if I have catalogue of ready (external) parts created in isolation, that I am simply putting together in an assembly...

> If you want to use binder to do `in context` modeling, then it is better to make an assembly that is dedicated for this purpose. 
> You then create a separate assembly in another file to use the model as part for assembling. 

Yet again, I am going to have to re-think my approach :+1: 
Not the first time .... part of the learning process I guess.

> The `freeze` function really only make sense if you use the part through external link. 

Understand - thank you
I just sometimes wish that all these wisdom and your advice 
found it's way into your https://github.com/realthunder/FreeCAD_assembly3/wiki
A lot of my confusion comes from religiously following some of it [image] [image]
I personally might be making progress in my understanding of the intricacies of Assembly 3 concepts,
but there must be others who would benefit from these clarifications?

closed 2022-01-11 by bambukouk