realthunder/FreeCAD_assembly3 #954: question, please - how do I "point to the link" in macro?

bambukouk, 2021-12-02:
Hi @realthunder this one is probably for you? 
can you help, please :+1: 

I have a macro that uses `dd` (Dynamic Data) to animate an assembly.
It works great if the animated sub-assembly is part of the active document, but...

if the sub-assembly I want to animate is linked externally, macro doesn't work with top level assembly because it cannot find the object it needs i.e.  `dd` - the only one available? is the `link to dd`

Normally the line in macro is:
`App.ActiveDocument.getObject('dd').ddCrank_angle= ang%360`

How do I change `('dd')`  to `('link to dd')` ??

I know that `link to dd` works fine, because I can use it and modify it manually. It's just the macro that is the problem...

realthunder, 2021-12-02:
With the selection path `sel = ('assy_C_top_level_','Assembly','Parts.Link002.Parts.dd.')`, use the following code to access the true object

```
obj = App.getDocument(sel[0]).getObject(sel[1]).getSubObject(sel[2]).getLinkedObject()
```

bambukouk, 2021-12-02:
Thank you very much indeed for your quick help!
It's probably me being dumb again...
but when I try the macro (modified as I understood your instructions?):

```
def float_range(first, last, step):
    while True:
        if step > 0 and first >= last+step/2:
            break
        elif step < 0 and first <= last+step/2:
            break
        yield first
        first += step

ang_b = 0
ang_e = -360
ang_s = -10
for ang in float_range(ang_b, ang_e, ang_s):
    App.ActiveDocument.getObject(App.getDocument(sel['assy_C_top_level_']).getObject(sel['Assembly']).getSubObject(sel['Parts.Link002.Parts.dd.']).getLinkedObject('dd')).ddCrank_angle= ang%360
    App.ActiveDocument.recompute()
    Gui.runCommand('asm3CmdQuickSolve',0)
    FreeCADGui.updateGui()
```
I get the following error message:

**class 'NameError'>: name 'sel' is not defined**

I will keep trying :+1:

bambukouk, 2021-12-02:
Thank you very much @realthunder for quick response.
Original macro (which works fine in the internal assembly) is:
```
def float_range(first, last, step):
    while True:
        if step > 0 and first >= last+step/2:
            break
        elif step < 0 and first <= last+step/2:
            break
        yield first
        first += step

ang_b = 0
ang_e = -360
ang_s = -10

for ang in float_range(ang_b, ang_e, ang_s):
    App.ActiveDocument.getObject('dd').ddCrank_angle= ang%360
    App.ActiveDocument.recompute()
    Gui.runCommand('asm3CmdQuickSolve',0)
    FreeCADGui.updateGui()
```

I have tried (to the best of my very limited ability) to modify it per your instructions for externally linked assembly...:

```
def float_range(first, last, step):
    while True:
        if step > 0 and first >= last+step/2:
            break
        elif step < 0 and first <= last+step/2:
            break
        yield first
        first += step

ang_b = 0
ang_e = -360
ang_s = -10

for ang in float_range(ang_b, ang_e, ang_s):
    sel = ('assy_C_top_level_','Assembly','Parts.Link002.Parts.dd.')
    obj = App.getDocument(sel[0]).getObject(sel[1]).getSubObject(sel[2]).getLinkedObject().ddCrank_angle= ang%360
    Gui.runCommand('asm3CmdQuickSolve',0)
    FreeCADGui.updateGui()
```

but it doesn't work, Clearly I am, doing something wrong...

Line:
`obj = App.getDocument(sel[0]).getObject(sel[1]).getSubObject(sel[2]).getLinkedObject().ddCrank_angle= ang%360`

is returning the following error message:
**class 'AttributeError'>: 'NoneType' object has no attribute 'getLinkedObject'**
changing it to:
`obj = App.getDocument(sel[0]).getObject(sel[1]).getSubObject(sel[2]).getLinkedObject('dd').ddCrank_angle= ang%360`
doesn't make any difference?

I will carry on trying :+1: 
but might need some more help, please....

realthunder, 2021-12-02:
Oops, sorry, forgot one parameter `retType=1`.

```
sel = ('assy_C_top_level_','Assembly','Parts.Link002.Parts.dd.')
obj = App.getDocument(sel[0]).getObject(sel[1]).getSubObject(sel[2], retType=1).getLinkedObject().ddCrank_angle= ang%360
```

bambukouk, 2021-12-02:
> Oops, sorry, forgot one parameter `retType=1`.
Don't know the difference - well above my abilities :-)
but delighted to report that it works :+1: 
Thank you again!!!!

closed 2021-12-02 by bambukouk