realthunder/FreeCAD_assembly3 #956: External Geometry issue

hamza00789, 2021-12-03:
https://user-images.githubusercontent.com/92737149/144589585-885b0c83-e170-4124-a7b4-6c9b27710e45.mp4

realthunder, 2021-12-07:
I can't reproduce the problem. Please post a file showing the problem.

hamza00789, 2021-12-07:
> I can't reproduce the problem. Please post a file showing the problem.

[image] 

[13.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7670409/13.zip)

hamza00789, 2021-12-16:
> I can't reproduce the problem. Please post a file showing the problem.

There is another similar problem, look at it. When I select a circle, the result is a dot, and when I select another circle, the result is the opposite. 

[image] 
[image] 
[image] 
[image] 
 
[1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7727615/1.zip)

closed 2022-04-29 by realthunder