realthunder/FreeCAD_assembly3 #959: GUI placement sphere not working

loops, 2021-12-03:
[knife2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7651073/knife2.zip)
----
OS: Fedora 34 (Workstation Edition) (sway/sway)
Word size of FreeCAD: 64-bit
Version: 0.20.30624 (Git)
Build type: Release
Branch: LinkDaily
Hash: de6bb54511e7cf3f5f50ebb732ea47980687a0a1
Python version: 3.9.7
Qt version: 5.15.2
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: English/Canada (en_CA)
----

If you open the attached file, and double-click on "Approximation_Curve (Mirror #1)"  the placement sphere appears around the part, but the control arrows are unresponsive to any clicks; so you can not change say the distance between the mirror halves.    Going to the property "Base -> Placement -> Position -> Z" and changing the value there moves the control sphere and mirror elements properly.  

Both methods of changing the mirror position work as expected in 0.19.

realthunder, 2021-12-06:
I can't reproduce the problem. Can you please make a screencast showing the problem?

loops, 2021-12-06:
Hmm.. i'm sorry to take your time on a potentially minor or  local config issue.  Maybe there is just something wrong with my setup. I've just recently switched to the Sway window environment, so I may have something misconfigured.

I will attach a screencast for what it's worth, but i'm fine to just close this issue for now until I can gather some more information.  The first attempt in the video is with 0.19, and the second attempt is with the current released AppImage.

https://user-images.githubusercontent.com/6493/144872241-3fb0f85f-c5fe-4fc8-9a29-72cdb12490d3.mp4

closed 2021-12-07 by realthunder