realthunder/FreeCAD_assembly3 #960: visibility problem

hamza00789, 2021-12-04:


realthunder, 2021-12-06:
This behavior is inherited from upstream, and it is a designed behavior. The visibility of the `Plain` group is used as a shortcut to toggle all its children's visibility together, while at the same time, each individual object can toggle its own visibility without affecting the parent group's visibility.

This behavior has its pros and cons. What you prefered is implemented in the `App::Part` container.

hamza00789, 2021-12-06:
Looks like you didn't understand me.  Or I could not understand you. If it
is designed behavior So how will it be determined that the group now has to
show all its children in this bar and in this bar show the child that is
already visible. Can you explain how it works?

realthunder, 2021-12-06:
When you hide the group, it will make a snapshot to remember what objects are visible at the time. Next time the group is made visible, it will show those objects. If you toggle individual object in the group while the group is hidden, it will adjust the snapshot for that object.

To toggle all children visibility of the group,  right click the object and select `Visibility -> Toggle group visibility`. I have just added code in the above referenced commit to support toggle children visibility by `Ctrl + Click` the eye icon of the group. There is also a fix of a bug in group visibility snapshot.

closed 2021-12-06 by hamza00789