realthunder/FreeCAD_assembly3 #961: Problem with the display of the toolbar view

lebearnais64, 2021-12-04:
Hello.
I move the toolbar  view and I put it on the edge of the screen. I use the mac os version.
Now it is impossible to move the toolbar.
[image]

I uninstall freecad and install il again. Still have the problem.

Thanks for your help

hamza00789, 2021-12-04:
> Hello. I move the toolbar view and I put it on the edge of the screen. I use the mac os version. Now it is impossible to move the toolbar. [image]
> 
> I uninstall freecad and install il again. Still have the problem.
> 
> Thanks for your help      
 
toolbar moving in only four sides. 
top bottom and left right  

https://user-images.githubusercontent.com/92737149/144728273-6b67a8a0-52d8-46f1-8e5d-8ba891a715c0.mp4

lebearnais64, 2021-12-05:
In fact I moved the toolbar but the toolbar did not plug on the left side as you show. I have two screen so the toolbar is limit to the 2 screens and I cannot have the cross to move it. 
[image]

lebearnais64, 2021-12-05:
https://user-images.githubusercontent.com/17858470/144758399-4e599c2f-be1e-420d-bd62-07bbc90dc895.mov

lebearnais64, 2021-12-05:
the left side is my second screen. I can move the freecad window as shown in the movie but impossible to move the toolbar.
Any idea?

hamza00789, 2021-12-05:
> the left side is my second screen. I can move the freecad window as shown in the movie but impossible to move the toolbar. Any idea?

Once you've removed the freecad setting, the toolbar will return to its original location

lebearnais64, 2021-12-05:
thanks a lot. Everything is ok now. Just to redo the settings ;-)

closed 2021-12-05 by lebearnais64