realthunder/FreeCAD_assembly3 #963: [Feature Request] New constraint proposal: Radial Alignment (Also creates a Polar Array when multiplied)

ToniTen, 2021-12-09:
In my field, I often have to place parts radially on cylinders, for example screws going in or out of hollow tubes, or grease nipples on large slewing bearings. Other CAD programs, such as Solidworks, solve this with a polar array, and rely on the first part being properly constrained to the appropriate hole. See [this example file](https://github.com/realthunder/FreeCAD_assembly3/files/7687150/example.zip) for a simplified case where this would be useful. I think that we can do better in FreeCAD and for this I propose this new constraint:

### Radial Alignment
#### Formed by:
- A line (from now on **L**)
- A plane (from now on **P**)

#### Constrains:
- **Rx** and **Ry**, so that the normal vector of the plane aligns with a perpendicular to **L** that passes through the origin point of **P**. Leaves all translations (**Tx**, **Ty**, and **Tz**) and **Rz** free.

#### Parameters:
- Rx and Ry offsets between the radial direction and the normal vector
- Number of occurrences (For constraint multiplication, from now on **N**)
- Angle (For constraint multiplication, from now on **alpha**).

#### How does constraint multiplication work with this constraint:
##### Prerequisites:
- The owner of the plane must be the first element of a link array
- **N** and **alpha** must be non-zero

First click on multiply constraint button will add **N**-1 links to the link array, and put them in a circle with it's center on **L**, and that passes through the origin point of **P**, with it's radius perpendicular to **L.** The new links are then positioned so that the angle between the first and the last link is alpha, and the angle between two consecutive elements is **alpha**/(**N**-1)
These new links will then be constrained by this constraint. All elements will use the same **Rx** and **Ry** offsets.
Second click on the multiply constraint button will expand the constraint as usual into an array of constraints for easy customization,

By itself, this constraint will ensure that the element is always oriented radially around a given axis, so it's possible to use a second constraint (such as points on a circle, or axial alignment) remove the rest of the DoF. For example, this could place a screw going into or out of a hole in a cylinder, while forcing it to be perpendicular to the surface of the cylinder. For example, this would have been extremely useful for solving the issue in [this forum post](https://forum.freecadweb.org/viewtopic.php?p=520773#p520773)

kevenwyld, 2021-12-10:
I'm sorry if this is a dumb question, I'm just trying to understand what you're asking because I have faced a similar situation.

Are you suggesting expanding the existing [Multiply constraint](https://github.com/realthunder/FreeCAD_assembly3/wiki/Constraints-and-Solvers#constraint-multiplication) function?  This currently doesn't work because it only tries to multiply with similar holes that are coplanar.  It would be pretty cool if this could be extended to be used in situations like your example where they are on the same face, but the face is not a flat plane.

ToniTen, 2021-12-13:
> EDIT: Yes, I see where you said it's an extension of it now. I think multiplying constraints for holes with matching radii on the same face might actually be more like "the real world" than constraining to a polar array based on a single central point. Your part could be a tube, a ring, a concave mirror, an egg, etc. and it would always (with some exceptions I suppose) work.

Indeed I am extending the concept of multiply constraint. Multiply constraint right now is awesome, but it's limited to plane coincident and axial alignment (unless it was added elsewhere and the documentation has not been updated). The reason I did it in this way is that most of the time, the surfaces where the parts must be placed are flat, not part of a single curved surface. If a different constraint would work here, by say selecting all circular faces with the same radius and whose normal vectors pointed towards the line that was selected, that would also work, and might be a better idea.

Note the definition of a "line" in Assembly3 (emphasis mine, from [this](https://github.com/realthunder/FreeCAD_assembly3/wiki/Constraints-and-Solvers#constraining-geometry-element) section of the wiki):
 
> - Line, (of which only the direction is relevant for constraining, not the end points)
>   - Linear edge;
>   - The **normal vector of the surface of a circular edge;**
>   - The **normal vector of the surface of a planar face;**
>   - The **revolving axis of a cylindrical face.**

Picking a linear edge to act as the line that defines the radial direction will be very rarely done. Instead, circular edges (cross section of a tube) or flat faces (cap on a tube) should be the most often geometry elements selected for this constraint

realthunder, 2021-12-24:
Not directly addressing the OP request, but the above commit adds support for using non planar edge or wire as element for constraint. asm3 will try to create a BSpline face using the wire, and use the face for constraining. The origin of the element is defined as the bound box center. The orientation is defined as the normal of the projected point from the bound box center. A quick demo would be that the circular hole on the cylinder surface (in [this](https://forum.freecadweb.org/viewtopic.php?p=520773#p520773) post) can be directly used for constrain.

I'll extend the support to constrain multiplication in the future.

realthunder, 2021-12-24:
In case anyone is wondering, you can use `G, G` shortcut to select higher level element, such as `Wire` here for constraining. This would be useful for example when constraining against a square hole.