realthunder/FreeCAD_assembly3 #964: Pocket to face

maxsnts, 2021-12-10:
Hello

I realize that the release notes say something about this 
"PartDesign: fix Pocket up to face, change algorithm for better topo naming"

But i keep having problems.
Here what happened to me mode than once.

How to replicate with 2021.10.15 daily on OSX (have not tried other OS)
- Create a Pad
- Create a sketch on one face
- Create a circle
- Pocket that circle with type "Up to face", selecting the opposite face
- Pocket is created on tree but we can not see it on model
- Delete Pocket on tree
- Pocket that circle with type "Up to face", selecting the opposite face adding an "offset to face" of 0.01
- Pocket is created on tree and we can see it on model
- Delete Pocket on tree
- Pocket that circle with type "Up to face", selecting the opposite face with NO offset
- Pocket is created on tree and we can see it on model
- Restart FreeCad, and this can be replicated again

Conclusion as strange as it sounds: 
Once the pocket it was created with the offset, it works correctly but once i restart freecad i can no longer pocket to face without adding an offset at least once.

Weird is it not?

On the weekend I will try to replicate the behavior on another MAC and see if it is the same.

Thank you very much for all you work.

realthunder, 2021-12-10:
Should be fixed by the above commit. It is indeed a quite bizarre problem. Thanks to your detailed steps to reproduce, I was able to locate the fix that is recently added to upstream, and made it to work here.

closed 2021-12-10 by realthunder

maxsnts, 2021-12-10:
Glad to help even if in this small way.
I don't know enough to help more because i just started learning 3D modeling.

I'm always in doubt if its my problem, that's why i make multiple tests and try to replicate to concrete steps.
(my developer background helps on that, i report problems as i wish they were reported to me)

Once again, thanks for you work. 
(I would help with Patreon, but i can't. sorry)