realthunder/FreeCAD_assembly3 #965: The array tool (draft WB) gives an unexpected (wrong) result when "Expand array" parameter is set to "True"

mwlandolsi, 2021-12-13:
Hello
Just create a circular array (Draft WB) of a cylinder (from Part wb) then set the "Expand array" parameter is set to "True". 

[image]

Now modify the "Radial Distance" for example (also the same problem with "Tangential Distance"). The result is an incorrect display of the array.

[image]
[image]

[image]

The good thing is that  The result of the boolean operation issued from the pattern is correct.

[image]
here is the file :
[cylinder patern and cut.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7703212/cylinder.patern.and.cut.FCStd.zip)

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-12-15:
One of the purpose to provide the option to expand the array is so that the user can manually move some of the array instances. So, the placement syncing is disabled by default if the array is expanded. If you still need placement sync while expanded, you can turn on property `AlwaysSyncPlacement`.

mwlandolsi, 2021-12-15:
Many thanks

closed 2021-12-15 by mwlandolsi