realthunder/FreeCAD_assembly3 #966: Constraint flipping, for a lack of a better description...

BloodyRain2k, 2021-12-14:
I'm gonna start directly with the images because I have no idea how to word this problem...
Initial state of the constraints:
[image]
[image]

State after a movement occurred and constraints recalculated into a different (but closer) possible orientation.
[image]
[image]
In this case the distance constraint flipped and was the thread profile sketch no longer keeping below the upper part.

File: [Soap_Stirrer.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7714266/Soap_Stirrer.zip)
If you want to reproduce what I did for the screens:
Lower the length of the "pad_Base" pad feature from 12 to 8 mm, that should cause the distance constraint to snap into the 2nd possible orientation.

I understand the underlying issue: when recalculating the new possible configuration to maintain all constraint rules, some like the distance constraint in this case, have more than one option, and they go for the shortest movement.
What I don't know is what would have to be done so the constraints can understand their "direction", for the lack of a better word in this case, so they only "move" but not flip entirely.

Like if I were to change the "pad_Base" length in small steps a few times, I would get to the intended result where it did not flip.
And the only two ways of getting there are either: not flipping the constraint by changing the pad length only in small steps until the new value is reached, or overshooting first past the new value (from 12 to 6 for example) and then to the actual intended value but at least enough so the constraint flips again (from 6 to 9 for example).

So yeah, I can't really offer a solution here and I'm not really expecting one at the time to begin with because this is a very complicated problem.
But maybe it can be kept around until someone can think of a way to preserve constraints well in such recalculation.

Or just point out to me how to design constraints so they are left with no options.
Although I think that currently FreeCAD doesn't like that kind of over-constraining...

bambukouk, 2021-12-15:
1. you have partially redundant constraints in `sk_Chuck_Base` sketch - no good :-1: 
2. solution to your problem is simple?
    don't constrain distance between two points (as you say - it has more than one solution)
    instead create vertical construction line between these two points and constrain the length of this line

Here is you base modified to 8mm with your sketch remaining as intended:
[image]

BloodyRain2k, 2021-12-15:
> 1. you have partially redundant constraints in `sk_Chuck_Base` sketch
How do I fix that? Because if I just delete it, the whole thing gains a degree of freedom and is enabled to shear...

> 2. I feel stupid for not realizing that that's the reason why there's 2 kinds of 'distance' constraints... Thanks <3

closed 2021-12-15 by BloodyRain2k

bambukouk, 2021-12-15:
>  1. you have partially redundant constraints in `sk_Chuck_Base` sketch
> How do I fix that? Because if I just delete it, the whole thing gains a degree of freedom and is enabled to shear...

you need to re-think how it is constrained
try... if you get stuck I will show example tomorrow

> I feel stupid for not realizing that that's the reason why there's 2 kinds of 'distance' constraints... 

 both types are needed (for different things)