realthunder/FreeCAD_assembly3 #968: [BUG] Pad creation leads to the loop

akshimassar, 2021-12-16:
[models.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7727505/models.zip)

During design I've encountered a situation where padding the sketch (fuse_7_relay_2_holder -> Body -> Sketch) leads to FC become unresponsive and consume 100% CPU. Thought it might be useful to provide an example for debugging.

OS: Arch Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28263 (Git)
Build type: Release
Branch: makepkg
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.9.7
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

akshimassar, 2021-12-16:
In fact it is a stable repro. I tried wiggling around to get rid of the issue but obvious steps like moving tip to the end or deleting boolean operation didn't help.

akshimassar, 2021-12-17:
I was able to reproduce the issue on a simpler example. It has to do with shape binder to the other file.
[image]
* create a new file (test), create a body here, simple sketch + pad
* create second file (test2), create body
* save both files
* while test2->body is active, select test->Body and click boolean operation, press ok (boolean references test->Body)
* add sketch to test2->Body
* try to pad the sketch
* FC falls to infinite loop

realthunder, 2021-12-19:
Should be fixed now. Thanks for reporting.

closed 2021-12-19 by realthunder