realthunder/FreeCAD_assembly3 #971: Path workbench does not work properly anymore

snelweg, 2021-12-17:
Using the latest stable linkstage version I notice that some Path wb functions do  not work anylonger.
For instance the pocket operations will not even show a view in the combo box.

I now have to design in your version, which I gladly do. But then to make gcode I have to switch to FC 0.20, which I hate to do. 
There are a lot of errors, like for example the multiple solids problem in regular FC. 

I do  not want to sound ungrateful or pushy but Is there any change of a merge so that the latest functions will be available?

Lately I have been using your spreadsheet functions a lot, I love them. Especialy the demonstration for a BOM that you made is very nice. Something FC has needed for a long time. I am busy making the BOM spreadsheet a little more usable by counting unique parts and displaying the number of parts in a column. This BOM is great for furniture makers and such.

Thanks very much for all your efforts, it is much appraciated.

zohozer, 2021-12-18:
Hi. Nice to see someone working to improve usability of FreeCAD for furniture design & fabrication. What is your username on FreeCAD forum? I am very interested in using FreeCAD for furniture production!

snelweg, 2021-12-19:
> Hi. Nice to see someone working to improve usability of FreeCAD for furniture design & fabrication. What is your username on FreeCAD forum? I am very interested in using FreeCAD for furniture production!

It is just an extension of Realthunders spreadsheet. Maybe some cvs export so that it can be used with Curtlist Optimezer. Not much time though.

realthunder, 2021-12-23:
Is it possible for you to try the Daily branch? This is the branch I'll try to keep in sync with upstream.

snelweg, 2021-12-23:
> Is it possible for you to try the Daily branch? This is the branch I'll try to keep in sync with upstream.

I switched over yesterday and I have to say that it works perfectly. No crashes and Path WB works properly

closed 2021-12-23 by snelweg