realthunder/FreeCAD_assembly3 #973: [BUG] Segfault on additive pipe when path is not touching profile

akshimassar, 2021-12-20:
* New document, new body
* Create sketch in YZ plane with square
* Create sketch in XZ plane with zig-zag path, at a distance from previous sketch
* Alternatively, open 
[test.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7743496/test.FCStd.zip)
* Create additive pipe of first sketch along second sketch
* Press ok / tick Update view
* Segfault

[image]

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3cda0) [0x7f32da36cda0]
#1  0x7f32d301abf0 in BRepFill_PipeShell::BuildHistory(BRepFill_Sweep const&) from /usr/lib/libTKBool.so.7+0x520
#2  0x7f32d3020ed1 in BRepFill_PipeShell::Build() from /usr/lib/libTKBool.so.7+0xbf1
#3  0x7f32d41f7f11 in BRepOffsetAPI_MakePipeShell::Build() from /usr/lib/libTKOffset.so.7+0x11
#4  0x7f32d256afa5 in BRepBuilderAPI_MakeShape::Shape() from /usr/lib/libTKTopAlgo.so.7+0x35
#5  0x7f32d4db680b in Part::TopoShape::makEShape(BRepOffsetAPI_MakePipeShell&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /usr/lib/freecad/lib/Part.so+0x6b
#6  0x7f324c17671f in PartDesign::Pipe::_execute(PartDesign::ProfileBased*, Part::TopoShape const&, int, Part::TopoShape const&, bool, int, Base::Vector3<double> const&, int, std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&) from /usr/lib/freecad/lib/_PartDesign.so+0x73f
#7  0x7f324c178cbc in PartDesign::Pipe::execute() from /usr/lib/freecad/lib/_PartDesign.so+0x44c
#8  0x7f32dc3fc53a in App::DocumentObject::recompute() from /usr/lib/freecad/lib/libFreeCADApp.so+0xaa
#9  0x7f32d4b9334e in Part::Feature::recompute() from /usr/lib/freecad/lib/Part.so+0xe
#10  0x7f32dc3698cd in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x22d
#11  0x7f32dc384504 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib/freecad/lib/libFreeCADApp.so+0x6c4
#12  0x7f32dc45e925 in App::DocumentPy::recompute(_object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x2e5
#13  0x7f32dc45eb5b in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x5b
#14  /usr/lib/libpython3.9.so.1.0(+0x148f08) [0x7f32dbb78f08]
#15  /usr/lib/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x2e3) [0x7f32dbb61333]
#16  /usr/lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x52a8) [0x7f32dbb5d218]
#17  /usr/lib/libpython3.9.so.1.0(+0x126fd9) [0x7f32dbb56fd9]
#18  /usr/lib/libpython3.9.so.1.0(_PyEval_EvalCodeWithName+0x51) [0x7f32dbb56c41]
#19  /usr/lib/libpython3.9.so.1.0(PyEval_EvalCode+0x23) [0x7f32dbc0dae3]
#20  /usr/lib/libpython3.9.so.1.0(+0x1ed9f4) [0x7f32dbc1d9f4]
#21  /usr/lib/libpython3.9.so.1.0(+0x1e96cb) [0x7f32dbc196cb]
#22  /usr/lib/libpython3.9.so.1.0(PyRun_StringFlags+0xa0) [0x7f32dbc12330]
#23  0x7f32dbf8b37e in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x6e
#24  0x7f32dce3551a in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x9a
#25  0x7f324838b067 in void Gui::_cmdDocument<char const (&) [12]>(Gui::Command::DoCmd_Type, App::Document const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const (&) [12]) from /usr/lib/freecad/lib/PartDesignGui.so+0x427
#26  0x7f324841c7ab in PartDesignGui::TaskDlgPipeParameters::accept() from /usr/lib/freecad/lib/PartDesignGui.so+0x9b
#27  0x7f32dd17727d in Gui::TaskView::TaskView::accept() from /usr/lib/freecad/lib/libFreeCADGui.so+0x6d
#28  /usr/lib/libQt5Core.so.5(+0x2f07a0) [0x7f32daa287a0]
#29  /usr/lib/libQt5Widgets.so.5(+0x2ff731) [0x7f32db677731]
#30  /usr/lib/libQt5Core.so.5(+0x2f07a0) [0x7f32daa287a0]
#31  0x7f32db5ca4e3 in QAbstractButton::clicked(bool) from /usr/lib/libQt5Widgets.so.5+0x43
#32  /usr/lib/libQt5Widgets.so.5(+0x25276c) [0x7f32db5ca76c]
#33  /usr/lib/libQt5Widgets.so.5(+0x254374) [0x7f32db5cc374]
#34  0x7f32db5cc595 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt5Widgets.so.5+0xf5
#35  0x7f32db5160be in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x20e
#36  0x7f32db4d2d62 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x82
#37  0x7f32db4daac9 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0xaa9
#38  0x7f32dcdc73a1 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x91
#39  0x7f32da9f13ba in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x13a
#40  0x7f32db4d957b in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt5Widgets.so.5+0x1bb
#41  /usr/lib/libQt5Widgets.so.5(+0x1b7a84) [0x7f32db52fa84]
#42  /usr/lib/libQt5Widgets.so.5(+0x1badb5) [0x7f32db532db5]
#43  0x7f32db4d2d62 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x82
#44  0x7f32dcdc73a1 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x91
#45  0x7f32da9f13ba in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x13a
#46  0x7f32dadd11f0 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt5Gui.so.5+0x6b0
#47  0x7f32dada67d5 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Gui.so.5+0xb5
#48  /usr/lib/libQt5XcbQpa.so.5(+0x62fcc) [0x7f32d0a92fcc]
#49  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x26c) [0x7f32d86f44dc]
#50  /usr/lib/libglib-2.0.so.0(+0xa8749) [0x7f32d8748749]
#51  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f32d86f1bc1]
#52  0x7f32daa49fe6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x66
#53  0x7f32da9efd2c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x12c
#54  0x7f32da9f8294 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x94
#55  0x7f32dcd16447 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x447
#56  freecad(+0x4928) [0x5594b0b6b928]
#57  /usr/lib/libc.so.6(__libc_start_main+0xd5) [0x7f32da357b25]
#58  freecad(+0x4d4e) [0x5594b0b6bd4e]
```

```
OS: Arch Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28263 (Git)
Build type: Release
Branch: makepkg
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.9.7
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)
```

realthunder, 2021-12-24:
I can confirm that it crashes with OCC 7.5, but 7.6 works, although the result is only correct with corner transition set to `Transform`. You can avoid the crash and obtain the correct result with 7.5 if one end of the vertices of the path sketch is inside the profile sketch wire.

[image]

akshimassar, 2022-01-28:
Thanks. I'm sorry for creating bug in the wrong project.

closed 2022-01-28 by akshimassar