realthunder/FreeCAD_assembly3 #974: LinkDaily ignores preference BaseApp/Preferences/Dialog/DontUseNativeDialog (Controls use of native dialogs) on Linux

ToniTen, 2021-12-20:
In upstream FreeCAD, we set BaseApp/Preferences/Dialog/DontUseNativeDialog to False to have FreeCAD use GTK system dialogs instead of the built in Qt ones. This is specially important for us for the following reasons:

- We use long file names and the native window is bigger and devotes more screen real estate to the file name column
- We work with network attached storage via gvfs (SSH) for CAD files, so they're accessible anywhere in the network, and well, to locate these directories, you need to go to /run/user/<UID>/gvfs/... every single time. These gvfs mounts appear on the sidebar of system open dialog straight away. We are copying the files to local directories for now, but it's a problem.
- On a related topic, the built in open dialog does not know about the system-wide bookmarks for specific directories. The native dialog does.

LinkDaily seems to ignore this preference. FreeCAD has supported this since 0.18 (according to [this](https://forum.freecadweb.org/viewtopic.php?f=10&t=30340&start=180#p273068)). and it's still working in the latest official 0.20 appimage. We understand that it's not available by default because it has caused issues to others, but we are surprised that it doesn't even work in LinkDaily

FreeCAD Version information
```
OS: Ubuntu 20.04.3 LTS (XFCE/xubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: Spanish/Spain (es_ES)
```

realthunder, 2021-12-23:
Are you sure upstream appimage is working? I tried 0.19 and the 0.20 weekly build appimage, neither are working. I think this is probably an appimage issue. Because it works if I run my build natively.

ToniTen, 2021-12-23:
Oh, sorry, I thought I had tested against the 0.20 Appimage, but I tested instead against the 0.20 from the FreeCAD Daily ppa. I have both, and have different launchers in my menu, but they are side by side.

I assume this is due to the fact that the Appimage bundles all the libraries instead of relying on system GTK/Qt. 

Thanks for your time and sorry for the confusion

closed 2021-12-23 by ToniTen