realthunder/FreeCAD_assembly3 #976: [FR] sketcher replace geometry

hamza00789, 2021-12-20:


realthunder, 2021-12-21:
It's already there. But it's not working in the latest release. I've already fixed it. Just select the two geometry you want to exchange and click the following command.

[image]

hamza00789, 2021-12-22:
> It's already there. But it's not working in the latest release. I've already fixed it. Just select the two geometry you want to exchange and click the following command.
> 
> [image]

But it is not working properly, it will work properly in the upcoming release

hamza00789, 2022-01-11:
thanks to solve

closed 2022-01-11 by hamza00789