realthunder/FreeCAD_assembly3 #981: 3D printing macro not pulling model out of the body container

HaasNL2, 2021-12-22:
I really like the 3D printing macro and use it a lot. But with Link3 it isn't able to pull the model out of the Body, it works just fine when the model isn't in a body. Probably super easy fix for anyone who actually knows what to look for. 
I know there are other macro's but this one somehow is the only one that I got to work, but if anyone has any better please let me know!
Below is the macro.
`import FreeCAD
import Mesh
import sys
import math
import os
import subprocess

SLICERAPP= "C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe"		 			# Put your Slicer program here

# some fuctions
def getPlacement(quat,vect,obj):
  if quat[3] > -1  and quat[3] < 1:
    delta = math.acos(quat[3])*2.0
    scale = math.sin(delta/2)
    rx = quat[0]/scale
    ry = quat[1]/scale
    rz = quat[2]/scale
  else:
    delta = 0
    rx = 0
    ry = 0
    rz = 1
  info0 = "translation "+str(vect.x)+" "+str(vect.y)+" "+str(vect.z)
  info1 = "rotation "+str(rx)+" "+str(ry)+" "+str(rz)+" "+str(delta)
  return info0+" "+info1
# some definitions
placement = App.Placement(App.Vector(0,0,0),App.Rotation(0,0,0,1))
# user need to set this directory where slicing software is located
OutDir = FreeCAD.ActiveDocument.FileName.replace(".FCStd", "--")
visible_objs = []
# Get Objects in document
doc = App.ActiveDocument
objs = doc.Objects
stlFile = ""
stlFiles = [ SLICERAPP ]
# hide all
for obj in objs:
   print(obj.Label + "//" + obj.TypeId)
   print(len(obj.InList))
   if obj.ViewObject.isVisible() and hasattr(obj, 'Shape') and (len(obj.InList) <= 1):
      visible_objs.append(obj)
for obj in visible_objs:
  stlFile = OutDir+str(obj.Label)+".stl"
  Mesh.export([obj],stlFile)
  stlFiles.append(stlFile)
  print ("Exporting " + stlFile + "\n")
print ("Calling subprocess: " + str(stlFiles)+"\n")
subprocess.Popen(stlFiles)`

realthunder, 2021-12-24:
I have modified the script. Please try. You can also selectively export objects by making selections in the tree view before invoking the macro.

[3d_Printer_Slicer_Individual_Parts_Modified.FCMacro.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7772617/3d_Printer_Slicer_Individual_Parts_Modified.FCMacro.zip)

HaasNL2, 2021-12-24:
Worked great thanks! Hopefully we can put this soon in  the newish plugin system!

closed 2021-12-24 by HaasNL2