realthunder/FreeCAD_assembly3 #986: [FR] Replace Face

hamza00789, 2021-12-23:
Replace face can make a specific face equal to a specific face or many faces can be made equal to a specific face. It works well in solid.   
As you can see in the picture.      

[image] 

[image] 
 
[image] 

[image] 

[image]

herrgerd, 2021-12-26:
Hey @hamza00789 I don't know if _you_ know, but @realthunder coding this branch of FreeCAD is a one-man-show!
When the bug-tracker is growing we should at least pause our feature-requests for a while.

He's doing all of this alone...

hamza00789, 2021-12-26:
Okay, now I won't be requesting any feature for a year. If a feature request is unbearable, turn it off

closed 2021-12-26 by hamza00789

ceremcem, 2021-12-27:
I can't talk *on* behalf of @realthunder but my own opinion about the feature requests as follows: 

* Rationale must be clearly explained. What is the benefit of that feature? How is it useful for a wide range of users?
* Usage (steps to perform that operation) must be clearly explained. 

Rest is all about triage (time, energy, motivation), which is all in @realthunder's jurisdiction. 

I guess the point @herrgerd underlines is that this project hasn't got enough resources (man power) and too many feature request won't find the chance to be implemented, thereby may demotivate the developers (@realthunder in this case).

hamza00789, 2021-12-27:
> I can't talk _on_ behalf of @realthunder but my own opinion about the feature requests is as follows:
> 
>     * Rationale must be clearly explained. What is the benefit of that feature? How is it useful for a wide range of users?
> 
>     * Usage (steps to perform that operation) must be clearly explained.
> 
> 
> Rest is all about triage (time, energy, motivation), which is all in @realthunder's jurisdiction.
> 
> I guess the point @herrgerd underlines is that this project hasn't got enough resources (manpower) and too many feature request won't find the chance to be implemented, thereby may demotivate the developers (@realthunder in this case). 

Wouldn't have requested it if it hadn't been useful.  Its benefit. If this is done with a Link this saves time. The work I have shown in advertisements can easily be done in any other way. Sometimes parts of the design are complicated. This can be very helpful if you want to get things done quickly in solid. Otherwise, the complex parts of the design just have to be converted into a surface then work on them, and then have to be converted back to solid. This can take a long time and can be difficult. I think it would be an exaggeration to say difficult but it takes more time. I wish I could give you an example. But for that, I have to do a lot of work and then I have to show you why it is necessary. These must be important. Just understand that. I will try to explain with a simple example (step file) why this is necessary. I used software called CATIA. When you are designing advanced level. So sometimes it feels necessary. At first glance, this may seem like a simple command, but when it is needed, its value is realized.