realthunder/FreeCAD_assembly3 #987: about set the spreadsheet alias using unicode

okkof2000, 2021-12-25:
when i used the  LinkDaily Branch 2021 1015, i found it can set a chinese as the alias,it is a great  function ，very thanks for you work 。but it is not work everytime，when i set alias by click right, try to set "顶面" as the alias ,by  the alias in "cell properties" ,the "ok" button is grey. 

when i try to set alias by python console , it will  raise "ValueError: Invalid alias"  like that
>>> App.ActiveDocument.getObject('Spreadsheet').setAlias('O12', '顶面')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ValueError: Invalid alias

As the sametime  ,try to set "底边" as alias，all the two ways are work.

when i  restart the freecad, all the chinese can not be set as the alias. -_- !

what can i do to keep it working all the time?

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Chinese/China (zh_CN)

realthunder, 2021-12-26:
I can't reproduce the problem. Maybe there is already an alias with the same name? Can you please post a file showing the problem?

okkof2000, 2021-12-27:
I record some videos ,and upload the videos and files to baiducloud
链接：https://pan.baidu.com/s/1aqxURerNXXi_IQTQEUGb_g 
提取码：1234

okkof2000, 2021-12-28:
i record some video and upload it
链接：https://pan.baidu.com/s/1WpM6M0xTUDJX2Am3otCLbQ 
提取码：1111

realthunder, 2021-12-30:
Yes, I can reproduce it in 1015 release, but not with a build from the current `HEAD`. Looks like it's been fixed. Please wait and try again with the next release. You can reopen this issue if you still have problem.

closed 2021-12-30 by realthunder