realthunder/FreeCAD_assembly3 #988: Macros not working

hamza00789, 2021-12-26:
They do not work properly in Link Daily. 

[image] 

Works well in 0.20 Version 

[image]

bambukouk, 2021-12-27:
> They do not work properly in Link Daily.

You could try to be a bit more descriptive?
I have plenty of macros that work perfectly fine in latest Daily version :+1: 

so, `NOT "Macros not working"`

I guess **`your issue is with Bevel macro specifically?`**

error message:
```
File "/home/owner/.FreeCAD/Macro/bevel.py", line 129, in execute
    FreeCAD/Console.PrintError("Error making boolean\n")
<class 'NameError'>: name 'Console' is not defined
```

hamza00789, 2021-12-27:
> > They do not work properly in Link Daily.
> 
> You could try to be a bit more descriptive? I have plenty of macros that work perfectly fine in latest Daily version 👍
> 
> so, `NOT "Macros not working"`
> 
> I guess **`your issue is with Bevel macro specifically?`**
> 
> error message:
> 
> ```
> File "/home/owner/.FreeCAD/Macro/bevel.py", line 129, in execute
>     FreeCAD/Console.PrintError("Error making boolean\n")
> <class 'NameError'>: name 'Console' is not defined
> ```

The first parametric defeaturing was also not working. But now it is working. But Bevel is still not working. Can you tell me why it is not working?

luzpaz, 2022-01-12:
Please retest on latest Daily and report back. Thank you!

hamza00789, 2022-01-12:
> Please retest on latest Daily and report back. Thank you!

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)  

It's still not working (bevel)

[image]

luzpaz, 2022-01-19:
x-posted https://forum.freecadweb.org/viewtopic.php?style=10&p=563017#p563017

mwganson, 2022-01-19:
There is a typo in the source code.  It should be FreeCAD.Console and not FreeCAD/Console.  I will fix it.

hamza00789, 2022-04-06:
> There is a typo in the source code. It should be FreeCAD.Console and not FreeCAD/Console. I will fix it.

The bevel macro is not working in the new release either.                   

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.402.26244 +4679 (Git)
Build type: Release
Branch: LinkDaily
Hash: 2ed36c0a234b3871a790ac8437f496f29e025ded
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

mwganson, 2022-04-06:
I don't have LinkDaily installed for testing.  Do you get any error message?  What are the steps to reproduce?  "It's not working," isn't much to go on.  Exactly how is it failing?

hamza00789, 2022-04-06:
https://user-images.githubusercontent.com/92737149/162018223-c55f2740-b7fd-46ea-ba30-1261441911e5.mp4     

[New Text Document.txt](https://github.com/realthunder/FreeCAD_assembly3/files/8428627/New.Text.Document.txt)

mwganson, 2022-04-06:
new version: 0.2022.04.06

Please test and report back.  You will need to restart FreeCAD after updating.

hamza00789, 2022-04-07:
> 

it still doesn't work. The error message is something like this.   

[Mp4.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8438930/Mp4.zip) 

[New Text Document.txt](https://github.com/realthunder/FreeCAD_assembly3/files/8438935/New.Text.Document.txt)

mwganson, 2022-04-07:
new version 0.2022.04.06b

hamza00789, 2022-04-07:
> 
Is working fine now.

closed 2022-04-07 by hamza00789