realthunder/FreeCAD_assembly3 #990: image - visibility (toggle show on top)

bambukouk, 2021-12-27:
When I insert image, I can control visibility (while I am in `image workbench`).
I use `visibility>toggle show on top`
This is perfect, combined with `transparency control in image appearance` I get what I need.
I can see both my model and underlaying image used for creation:
[image]

But.... in this condition (i.e. "show image on top") I loose control of visibility.
Even with visibility switched off, the image remains visible:

[image]

Plus... if I save the file with `visibility set to show on top`, it all goes back to default next time I open the file :-1:  and I have to change visibility on top again, and I can only do it from `image workbench` for this particular session.

[image]

I personally think that `visibility toggled to show on top` 
should remain so `until I toggle it OFF `?

i.e. - it should be like a transparency setting - set until changed. 
Not a temporary setting for just one session.

I attach file for experimenting with (as usual - rename to FCStd to use):

[visibility(image).zip](https://github.com/realthunder/FreeCAD_assembly3/files/7780117/visibility.image.zip)

closed 2021-12-31 by realthunder

realthunder, 2021-12-31:
Just submit a few follow up patches to fix a few glitches of on top rendering thanks to your sample file. Anyway, after considering a few factors, I decide to maintain the current behavior of on top control. That is, once an object is put on top, normal visibility control will have no effect. You are supposed to toggle on top group using `Ctrl + Shift + Space`, or `Alt + click` the eye icon. 

I have also add some visual feedback for on top display to the eye icon.

[image]

bambukouk, 2021-12-31:
Thank you!
Main thing was visibility on top
When will I be able to use it and test it? 👍 
I guess you are already in 2022 - have a great one!