realthunder/FreeCAD_assembly3 #991: polar pattern - number of occurrences limited to two digits only

bambukouk, 2021-12-28:
Unable to enter more than 99 occurrences (works fine in master branch - tested in 0.20)
but not in my latest daily:
```
OS: elementary OS 6.1 JÃ³lnir (Pantheon/pantheon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United Kingdom (en_GB)
```

[image]

mwlandolsi, 2021-12-28:
You can use "Formula Editor". When you hover the mouse pointer over the "Occurrences" filed, just click on formula. You can put more than 99.
This looks for me logical, because creating a big number of occurrences, causes some computers to become very slow.

bambukouk, 2021-12-28:
There is something very odd about it.
Yes, you are right that I can enter numbers larger than 99 by using expression, but...
it doesn't change the number in `Task View` (it still remains 99, regardless of the number entered in expression) it only updates correctly in `Property View`
And... why should entering this number as an expression rather than directly should help with "some computers becoming very slow" - haven't got the clue :-)
So it kind of works but leaves me rather puzzled...
Thank you anyway ;-)

realthunder, 2021-12-29:
This problem is fixed by upstream. And the fix has been merged here, so it will be available in next release.

closed 2021-12-29 by realthunder

bambukouk, 2021-12-29:
Thank you again for finding time for my trivial issues in the middle of all the work you doing merging things.
We appreciate your effort and look forward to the next release.
Looking forward to a great 2022