realthunder/FreeCAD_assembly3 #992: 'Part.Shape' object has no attribute 'countSubElements'

xiaomingneo, 2021-12-29:
I'm trying the simple example with just a Cube and a Cylinder,

while press the <Place Coincidence> button, the problem below occurs:

```python
10:42:30  Traceback (most recent call last):
  File "/home/neo/.FreeCAD/Mod/FreeCAD_assembly3/freecad/asm3/assembly.py", line 802, in execute
    fcount = shape.countSubElements('Face')
<class 'AttributeError'>: 'Part.Shape' object has no attribute 'countSubElements'
```

I'm using the  realthunder fork of FreeCAD:
FreeCAD-asm3-Daily-Conda-Py3-Qt5-20211015-glibc2.12-x86_64.AppImage

realthunder, 2021-12-29:
Fixed in the above commit. Please update you local clone of asm3 repository.

closed 2021-12-29 by realthunder

xiaomingneo, 2021-12-30:
it works.

thank you very much!

> Fixed in the above commit. Please update you local clone of asm3 repository.

commit 8e8977cb (2021-12-29): assembly: fix for backward compatibility

Fixes #992