realthunder/FreeCAD_assembly3 #993: Segmentation fault when start scetch

TarekMSayed, 2021-12-30:
OS: Ubuntu 20.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +3962 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

Here are the steps to get the crash:
1- part design
2- new
3- save as: test
4- add Part
5- add Body
6- Create sketch 
7- choose Object's XY
8- OK

then here is the crash

/tmp/.mount_FreeCAw8NgcG
FreeCAD 0.19, Libs: 2021.1015R24301 +3962 (Git)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f25bf457210]
#1  0x7f2565d1cefc in PartGui::ViewProvider2DObjectGrid::createGrid() from /tmp/.mount_FreeCAw8NgcG/usr/lib64/./PartGui.so+0x6cc
#2  0x7f2565d1dc19 in PartGui::ViewProvider2DObjectGrid::setEdit(int) from /tmp/.mount_FreeCAw8NgcG/usr/lib64/./PartGui.so+0x39
#3  0x7f25658ad0ce in SketcherGui::ViewProviderSketch::setEdit(int) from /tmp/.mount_FreeCAw8NgcG/usr/lib64/SketcherGui.so+0x53e
#4  0x7f25c1a49e84 in Gui::ViewProvider::startEditing(int) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0x14
#5  0x7f25c1a705b5 in Gui::ViewProviderDragger::startEditing(int) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0x15
#6  0x7f25c1602a98 in Gui::Document::setEdit(Gui::ViewProvider*, int, char const*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0xb98
#7  0x7f25c16112f9 in Gui::DocumentPy::setEdit(_object*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0xd9
#8  0x7f25c16114b7 in Gui::DocumentPy::staticCallback_setEdit(_object*, _object*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0x57
#9  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(+0x1651f8) [0x7f25c0f111f8]
#10  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x2df) [0x7f25c0f0c94f]
#11  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x5104) [0x7f25c0f72474]
#12  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(+0x184d19) [0x7f25c0f30d19]
#13  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalCodeWithName+0x51) [0x7f25c0fd6f81]
#14  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(PyEval_EvalCodeEx+0x3a) [0x7f25c0fd6fca]
#15  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(PyEval_EvalCode+0x1c) [0x7f25c0fd6fec]
#16  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(+0x22b09f) [0x7f25c0fd709f]
#17  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(+0x258bbc) [0x7f25c1004bbc]
#18  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libpython3.9.so.1.0(PyRun_StringFlags+0xa7) [0x7f25c1004c87]
#19  0x7f25c067d46c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADBase.so+0x6c
#20  0x7f25c1694091 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0x91
#21  0x7f25c16bdf3b in void Gui::_cmdDocument<std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> > >(Gui::Command::DoCmd_Type, App::Document const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >&&) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0x3cb
#22  0x7f2565d84654 in PartGui::TaskAttacher::~TaskAttacher() from /tmp/.mount_FreeCAw8NgcG/usr/lib64/./PartGui.so+0x444
#23  0x7f2565d84a2a in PartGui::TaskAttacher::~TaskAttacher() from /tmp/.mount_FreeCAw8NgcG/usr/lib64/./PartGui.so+0xa
#24  0x7f25c19a7a5a in Gui::TaskView::TaskDialog::~TaskDialog() from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0x3a
#25  0x7f2565d717da in PartGui::TaskDlgAttacher::~TaskDlgAttacher() from /tmp/.mount_FreeCAw8NgcG/usr/lib64/./PartGui.so+0xa
#26  0x7f25bf99b54e in QObject::event(QEvent*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libQt5Core.so.5+0x76
#27  0x7f25c01ae7f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#28  0x7f25c01b431d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#29  0x7f25c1642849 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0x89
#30  0x7f25bf986606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libQt5Core.so.5+0x86
#31  0x7f25bf98683e in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libQt5Core.so.5+0x1da
#32  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libQt5Core.so.5(+0x1dc54c) [0x7f25bf9b554c]
#33  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f25bbfee1ed]
#34  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/./libglib-2.0.so.0(+0x55481) [0x7f25bbfee481]
#35  /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f25bbfee521]
#36  0x7f25bf9b50e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libQt5Core.so.5+0x5e
#37  0x7f25bf983019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libQt5Core.so.5+0x187
#38  0x7f25bf986e8f in QCoreApplication::exec() from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib/libQt5Core.so.5+0xfd
#39  0x7f25c15b8b43 in Gui::Application::runApplication() from /tmp/.mount_FreeCAw8NgcG/usr/bin/../lib64/libFreeCADGui.so+0x413
#40  /tmp/.mount_FreeCAw8NgcG/usr/bin/FreeCADLink(+0x448d) [0x55ee4086948d]
#41  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f25bf4380b3]
#42  /tmp/.mount_FreeCAw8NgcG/usr/bin/FreeCADLink(+0x47e9) [0x55ee408697e9]

TarekMSayed, 2021-12-30:
Update: 
the same issue happened with FreeCAD original AppImage stable and daily build also with daily build install using apt.
Finally removing the check from the `show grid` box at `Edit>Prefrences>Sketcher tab` Fixed the issue

bambukouk, 2021-12-30:
It could posdibly be Ubuntu related?
I just tried appimage of latest release of 0.20 (I know it is not link branch) and it crashes on starting with remarkably similar segmentation fault message. 
Common denominator is OS in my case Elementary (Ubuntu based)
??