realthunder/FreeCAD_assembly3 #998: Arch Wall and Window does not move correctly along with Arch Floor

cadmarx, 2021-12-31:
I am using FreeCAD_assembly 3 version 2021.10.15. It seems Arch Wall and Window does not move correctly according to Arch Floor coordinates; the problem is even more apparent when a change in coordinate of Arch Floor is UNDONE. I have attached a GIF of the operations for clarity.
[image]

luzpaz, 2022-01-12:
Can you reproduce in upstream master v0.20dev?

cadmarx, 2022-01-18:
Will do soon..

cadmarx, 2022-01-22:
Hi luzpaz,

I checked with FreeCAD_0.20.26858. It has the very same problem.

luzpaz, 2022-01-22:
Close....please open a ticket on the forum. Lets let realthunder continue fixing assmebly3 and porting his changes upstream. There are other arch devs that can tackle this.

closed 2022-01-22 by luzpaz