realthunder/FreeCAD_assembly3 #999: SubtractiveHelix creates weird shell

marthinwurer, 2022-01-02:
[image]
[Helix_tube.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7798495/Helix_tube.zip)

I attempted to make threads around the end of a pipe, and FreeCAD threw a blob in there that clogged the pipe.

bambukouk, 2022-01-02:
Try the attached file
(I just renamed it.zip rather than compress already compressed file - to open in FreeCAD just rename it to .FCStd)
[Helix_tube(BAMBUKO).zip](https://github.com/realthunder/FreeCAD_assembly3/files/7799039/Helix_tube.BAMBUKO.zip)

[image]

marthinwurer, 2022-01-04:
hmm, that has the bottom edge flat but I wanted it to be chamfered. I guess I could try chamfering it after the fact.

bambukouk, 2022-01-04:
Simply because it was quicker :-)
The point I was making - there were problems with your sketch.
I am sure it can be chamfered both ends - just need to get the sketch right
I leave it to you ;-)

luzpaz, 2022-01-12:
Is this ticket then PEBKAC ?

bambukouk, 2022-01-12:
translate please - what is PEBKAC?

luzpaz, 2022-01-12:
PEBKAC (an acronym for "problem exists between keyboard and chair") :wink:

bambukouk, 2022-01-12:
That's new one to me 🙂
People need some place to ask for help...
Here in "issues" is  possibly not the best place?
but they don't want us on FreeCAD forum either...🥸
unless it is in assembly section?

luzpaz, 2022-01-12:
Yea, it's an issue since RT made his own branch. Probably a room on Discord or something? I'm closing this ticket

closed 2022-01-12 by luzpaz

bambukouk, 2022-01-12:
I joined Discord to see what's it like .... 
not for me (maybe I am just old 🤣)

luzpaz, 2022-01-12:
There is also https://gitter.im/FreeCAD/FreeCAD

bambukouk, 2022-01-12:
To me it's like another Discord 😬
Personally I like forum with all the functionality it offers
Pity they can't create section dedicated to Link branch....