realthunder/FreeCAD_assembly3 #1000: [Question] Increase pie menu item count

ngocminhdao88, 2022-01-03:
I want to map my often used workbenches to a pie menu. Some how the item count is capped at 8. Can i increase this number?

Another question. I want to hide text, only show icon on some pie menu. Is this possible?

ngocminhdao88, 2022-01-06:
Update with picture, because one pic worth than thousand words.

I want to have more items in my custom pie-menu
[image]

But only first 8 are showed. I also want to hide the item's text in some pie-menu but not all.
[image]

I am using the latest Link Branch release

> OS: Manjaro Linux (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

luzpaz, 2022-01-11:
Relevant areas in the code:  
* https://github.com/realthunder/FreeCAD/blob/LinkDaily/src/Gui/PieMenu.h
* https://github.com/realthunder/FreeCAD/blob/LinkDaily/src/Gui/PieMenu.cpp
* https://github.com/realthunder/FreeCAD/blob/aa5c706ef6d0ac5b7e906084a83476e7823ea7d5/src/Gui/ViewParams.h#L294-L312