realthunder/FreeCAD_assembly3 #1002: how to un-link "linked files"

bambukouk, 2022-01-05:
I have a file that is externally linked to an assembly.

I want to use the same file in another totally different design.
Copy it to another project folder and when I open it all subshape binders inside it are gone wrong :-1: 
and error message complains about file exception "File it was originally linked to does not exist"

How do I, please tell FreeCAD that this is not any longer linked to anything?

ngocminhdao88, 2022-01-06:
If i understand you correctly, then you can unlink the subshapebinder by changing it's bind mode to frozen or detached.

[image]

A good practice is to freeze your design when you are done with your project or reach some mile stone.

bambukouk, 2022-01-06:
Thank you, but.... possibly you misunderstand me ( I guess it is not easy to explain well - my fault)
I will try again :+1: :
1. I have a `"File_A"` inside of which there are some sub-assemblies using  subshapebinders in part design.
2. I link externally some of these sub-assemblies to `"File_B"`
3. All is well until I want to use `"File_A"` in another design....
4. I copy `"File_A"` into this new design folder, but subshapebinders stop working because they are now supposedly bound to `"File_B"` 

 Why??? 
These subshapebinders are internal, binding only components inside `"File_A"` and nothing to do with `"File_B"`
It seems that linking them changes something?

Anyway, I don't want to unlink them inside `"File_A"` 
I need to disentangle the assumed linking to `"File_B"` - that is my problem.
Like I said in my original post I can handle it by re-associating correct components into affected subshapebinders,
but it is a bit of a PITA

I guess I will need to read up on `"Bind Modes"`...?

bambukouk, 2022-01-06:
Some more explanations, this time showing pictures:
1st is side by side view of the same file - on the left as it was created, on the right after it was used in externally linked assembly,
and then separated/copied into another folder to use in another assembly.
[image]

I am comparing one and the same subshapebinder (you can also see it in the file attached to my first post)

If I select affected shapebinder (on the right) it shows following info:
[image]

You can clearly see that it remains linked to another file(01_Assembly.FCStd) and that is what I am talking about.

1. Why does this only affect subshapebinders only? (after all it is assemblies consisting of many parts that are linked not just subshapebinders)
2. Assuming it is just my ignorance - what is the best way of dealing with it, please?
3. 
and @ngocminhdao88 you cannot unlink it by `changing the mode to frozen or detached` - it simply destroys the binding to original object it has ben bound to :-1:

closed 2022-01-08 by realthunder

bambukouk, 2022-01-08:
So I am guessing it was the real thing :+1:  
rather than me messing things up!! as is often the case...;-)
Thank you again @realthunder 

Hopefully this will make things easier in future

realthunder, 2022-01-08:
Just some background about this issue. There is a hidden property `Context` in `SubShapeBinder` to store the context object path of this binder when you last add some binding. This is what's allow the binder to auto update its bound object's relative placement when the binder's parent's placement changes. The problem you've encountered is due to adding a new binding in an external binder, in which case, its `Context` may be pointing to an external parent (relative to the binder). That's why it reports error when you move the file.

This shouldn't be considered as an error. And the patch here just silence the error report. Do note that if you are using the same  binder in different contexts (i.e. from different parents, which is generally not advisable), you may loose auto placement sync when one of the parent changes. Copying the file is fine, but you'll probably need to re-establish the context when using the file in new context. In short, if you see the binder (i.e. the bound shape) is out of place, just double click the binder to re-sync.

There is an old bug (which has been fixed already) that makes the problem mentioned here worse. By right, the binder is supposed to resolve the context object path so that it points to the last non-common ancestor of the binder relative to the bound object. This will reduce the need to re-establish the context in many use cases.

bambukouk, 2022-01-08:
Thank you for detailed explanation 👍
Not quite sure that I understand  `using the same binder in different contexts`
but maybe it is because I am trying to keep things simple (as the saying goes... KISS)
I guess I will have a chance to test it all, once next release is available.

Thank you again for all your work!