realthunder/FreeCAD_assembly3 #1004: double mouse click to rename

bambukouk, 2022-01-09:
I have long "complained" about issues I (and others) have with renaming things.
https://github.com/realthunder/FreeCAD_assembly3/issues/614
This is definitely not a mouse problem because using F2 is exactly the same (so both mouse and keyboard have the same issue).
I have even change both my mouse and keyboard to "better quality" ones with no change...

Now, I have just noticed that `double clicking` constraints name does straight to `"rename"` and is 100% reliable.
This is great and would be desirable for everything else in tree view, unfortunately double clicking item name has different meaning depending what it is:
constraints (list) - rename
constraints (individual) - rename
assemblies - rename
assembly parts - rename
elements (list) - rename
elements (individual) - transform
Above is just Assembly3 components. 

(I will not even bother going into the rest of mainstream FreeCAD it's a mess of conflicting actions...)

That's great and it solves the problem at least for Assembly3, apart from the `inconsistency` of `individual elements` where double click goes to `transform`.

May I, please suggest that all Assembly3 components have the same action if one double clicks on the name of component?
i.e. double clicking the name of individual element should go directly to rename and transform should be right click context menu option.

bambukouk, 2022-01-11:
Just to expand and explain my comments...
This is something like what I had in mind:
[image]

Rabbit-sk, 2022-01-11:
Hi @bambukouk !

I think that single left click on item icon should not expand or collapse tree branch, because there are already arrows for that. 
Selection of an item would not be possible by clicking on its icon. Such behavior would not be intuitive and is a significant change from the current way in which item selecting is working now and it would be undesirable for many users.

bambukouk, 2022-01-11:
@Rabbit-sk 
Thank  you,

> I think that single left click on item icon should not expand or collapse tree branch, because there are already arrows for that. 

but unfortunately this behaviour already exist - try it :+1: 

> Selection of an item would not be possible by clicking on its icon. 

it is not possible at the moment indeed

> Such behavior would not be intuitive and is a significant change from the current way in which item selecting and expanding is working now and it would be undesirable for many users.

Yes, I know there is always vocal bunch of guys who will fight to the bitter end to keep things as they are... :-)

Rabbit-sk, 2022-01-11:
>>I think that single left click on item icon should not expand or collapse tree branch, because there are already arrows for that.

>but unfortunately this behaviour already exist - try it 👍

>>Selection of an item would not be possible by clicking on its icon.

>it is not possible at the moment indeed

There may have been a misunderstanding on my or your side because, as seen in the following video, a single left-click on an item icon does not expand the item, but selects it. 

https://user-images.githubusercontent.com/74265320/149016892-2d24dd2f-d996-4ed4-b5c1-5e422d670721.mp4

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

bambukouk, 2022-01-11:
@Rabbit-sk 
> There may have been a misunderstanding on my or your side because, as seen in the following video, a single left-click on an item icon does not expand the item, but selects it.

The action depends on what item you are clicking on.
If you look around FreeCAD almost every item has different action (I guess depending on what specific developer of this one little bit bit of FreeCAD felt was the best way :-1: )

It's a mess. OK, I like everybody else learned to live with it, but that doesn't mean this is the way we should be doing it....
With @realthunder doing his own thing anyway, and being that he is a single guys with just one voice and one opinion, there is some chance for sanity :+1: 

So, by all means - let's argue how it should be done, but whatever the choice is let it be consistent :+1:

bambukouk, 2022-01-12:
now that we are merged with 0.20 this applies as well@
https://wiki.freecadweb.org/Std_UserEditMode