realthunder/FreeCAD_assembly3 #1005: Offset Issue

hamza00789, 2022-01-09:
The offset command intersection option is not working properly  

[image]   
 
https://user-images.githubusercontent.com/92737149/148694002-bb66dcf3-e143-4416-83fa-572dbdb272c6.mp4

luzpaz, 2022-01-12:
Please always post your full [About](https://wiki.freecad.org/About) info. Thanks

hamza00789, 2022-01-12:
> Please always post your full [About](https://wiki.freecad.org/About) info. Thanks
 
old version [FreeCAD-asm3-Daily-Win64-Py3-Qt5-20211015.7z] did not work          
 
It did not work in the old version but it works well in the new version  
 
the new version [FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220111.7z]  works well

hamza00789, 2022-02-12:
The problem described above was resolved for this file. Now I have this problem again for another file.