realthunder/FreeCAD_assembly3 #1006: [bug] Freecad crashes when missing thumbnails in subfolders

HaasNL2, 2022-01-10:
I have noticed when I have a sub-folder with unloaded files Freecad crashes to desktop.
I was trying to find out what was causing the crash, and then I opened the folder with the files in windows explorer and let it load all the thumbnails and it stopped crashing instantly. 
The folder on which it crashed had one .obj file the rest where Solid-edge files.

`OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Dutch/Netherlands (nl_NL)
`

realthunder, 2022-01-11:
Can you please post a 'subfolder' causing the problem? By crash, do you mean a hard 'Segment fault' kind of crash?

HaasNL2, 2022-01-11:
> Can you please post a 'subfolder' causing the problem? By crash, do you mean a hard 'Segment fault' kind of crash?

First, thanks for the new release!!!
Second, it just freezes for a second and exits. Not sure if there are any exit logs I can give?
The folder is also super random it seems to happen very random but if I go true some documents folders it crashes very quickly. 
The only thing that is pretty much happens every time is when it crashes and then I open the folder in explorer let it load all the thumbnails, then go back to Freecad it doesn't crash anymore.

realthunder, 2022-01-12:
Do you mean it crashed while you were browsing files using the open file dialog? Are you using the system file dialog or the Qt one? Can you try toggle the following option and see if there is any difference?

[image]

HaasNL2, 2022-01-12:
Not using native really stops it from crashing so thanks for that tip! I noticed this crash also happens in the main branch.
Should probably either disable the native dialog or fix it tho. 
But I find it a bit weird that there is no bug report already on main branch, either nobody uses the native window or there is something on my system causing the crash. Do you think it is wise to open a bug report on the main branch bug tracker?

realthunder, 2022-01-12:
Yes sure, you can make a post in the forum describing the problem to see if there are others having similar issue. I suspect it is something on your system that's causing problems.

luzpaz, 2022-06-30:
@MisterMakerNL is this still an issue?

MisterMakerNL, 2022-07-01:
I tested it again and it doesn't crash anymore. 
I installed windows PowerToys that does stuff with the thumbnails, although it added support for svg and gcode. Maybe that fixed it? 
Tested on
`[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Assembly3 0.11.3
  * Behave-Dark-Colors 0.0.1
  * BIM 2021.12.0
  * FeedsAndSpeeds 0.4.0
  * FEM_FrontISTR 0.1.0
  * IconThemes
  * lattice2 1.0.0
  * ProDarkThemePreferencePack 1.0.0
  * SelectorToolbar
  * sheetmetal 0.2.49
  * ToolbarStyle
[/code]
`

closed 2022-07-01 by MisterMakerNL