realthunder/FreeCAD_assembly3 #1008: Synta Error to create any sketch.

QuakeString, 2022-01-11:
Hi I'm trying to create a sketch it always showing syntax error. I tired to make a body and then create a sketch, or in bim wb, sketchh on a face or on Sketcher WB just create a sketch also "syntax error"
This problem I'm seeing from yesterday, I was on 2nd last version of Freecad now I updated to freecad Stable 2022-01-11
also tried the daily branch too, has same issue.
here is detail
OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
Daily Branch
[image]
Stable Branch
[image]

bambukouk, 2022-01-11:
Just tested on mine:
```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```
no problem at all... ??

QuakeString, 2022-01-11:
> Just tested on mine:
>
> ```
> OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
> Word size of FreeCAD: 64-bit
> Version: 2022.111.30764 (Git) AppImage
> Build type: Release
> Branch: LinkDaily
> Hash: [...]

I'm not sure but seems your and mine version is not same. 28429 | 30764

realthunder, 2022-01-11:
Maybe you have some external module that cause the error? Try rename `~/.FreeCAD/Mod` and see.

QuakeString, 2022-01-11:
I've completely remove .FreeCAD folder and installed BIM WB , now I can create sketchs but, some errors still there.
[image]

luzpaz, 2022-01-13:
Looks like you need to install the solvespace solver

QuakeString, 2022-01-13:
> Looks like you need to install the solvespace solver

I've installed solvespace and then open free cad, but still has same message, 
[installed solvespace from AUR, do i need to do any more configuration on  FreeCAD? such as path to solvespace ?]

realthunder, 2022-01-13:
I think you have installed asm3 workbench in your FreeCAD user directory (~/.FreeCAD/Mod). If so, please go to the directory and run `git pull` to update. After that it will prompt from insalling SolveSpace Python binding, when you activate asm3 workbench.  Installing it through your system Python may or maynot work depending on your system Python version, which is likely not the same as the one bundled in the release image.

QuakeString, 2022-01-13:
> 

when I first time try to select Assembly 3 WB, it asked me to download and install slovespace, after I acknowledge with "Yes" it try to install automatically, and it out put a failed to install. though it output success on python command, and i restart FreeCAD, now there is no error of asm3.main.

luzpaz, 2022-04-11:
Issue is therefore solved?

QuakeString, 2022-07-07:
Yes, it is solved.

closed 2022-07-07 by QuakeString