realthunder/FreeCAD_assembly3 #1009: (just for info) cannot run newest version 2022.01.11 daily

bambukouk, 2022-01-11:
when trying newest daily release got this:

```
bambuko@Z97MD3H900631f2:~/Applications/FreeCAD_link_daily/link_daily_newest_appimage$ ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220111-glibc2.12-x86_64.AppImage
/tmp/.mount_FreeCA66xfSj
/tmp/.mount_FreeCA66xfSj/usr/bin/FreeCADLink: error while loading shared libraries: libOpenGL.so.0: cannot open shared object file: No such file or directory
```
solved very quickly by:
`sudo apt install libopengl0 -y`

```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)

```

still have an issue with  a failure to install py-slvs but that is another story....

realthunder, 2022-01-11:
Any error message when installing py-slvs?

bambukouk, 2022-01-11:
no just brief window saying that it failed...
have run FreeCAD latest version since and assembly3  runs fine with no complaints?
will keep you posted

StudioPetrikas, 2022-01-11:
I had a few issues with py-slvs, I've got them fixed. But here's what happened:

- I had Assembly3 folder in 'user' FreeCAD/Mod folder (main issue, it didn't proc the prompt to install py-slvs until I removed (renamed) that folder.
- I installed py-slvs on my global Python install, whereas I should've installed it on FreeCAD/bin/python.exe

Fixed these two issues, and it's now working flawlessly.

bambukouk, 2022-01-11:
```
    got the 'install py-slvs' prompt, agreed and got an error. Restarted FreeCAD Link.
After the restart it's working without issues.
```

Yeah, that sounds similar to my experience

mwlandolsi, 2022-01-11:
On Linux mint xfce 20.2 it does not run. Not tested yet on mint cinnamon. I double-click on .appimage file and nothing happens. I do not know where to find message errors if there are any!
None of the suggested propositions above tested because I do not know if it could be good solution for my OS or not.

bambukouk, 2022-01-11:
run FreeCAD from terminal and you will get error messages which will confirm what the problem is

luzpaz, 2022-06-30:
@bambukouk is it OK to close this ?

bambukouk, 2022-06-30:
sorry I thought it was closed?

closed 2022-06-30 by bambukouk