realthunder/FreeCAD_assembly3 #1017: Not possible to create fillet which would share edge with another fillet

Rabbit-sk, 2022-01-11:
Hi RealThunder!

I tried to make two fillets with the same radius of 500 mm, one next to the other, to round the edge of a 1000 mm thick object.

https://user-images.githubusercontent.com/74265320/149010478-f7a7a448-1b66-4619-b8bc-864e41b6c3bb.mp4

Output of the Report view:
```
20:30:10  3979.69 <App> Document.cpp(4262): Failed to recompute Unnamed#Fillet001: BRep_API: command not done
20:30:13  3982.38 <App> Document.cpp(4262): Failed to recompute Unnamed#Fillet001: BRep_API: command not done
20:30:13  Recompute failed! Please check report view.
```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)

obelisk79, 2022-01-12:
Just wanted to add the note that this issue exists in all versions of FreeCAD, it isn't exclusive to these builds.

herrgerd, 2022-01-13:
Yeah, it has something to do with the internals of the renderer or something. If you have 10mm plat with two radii of 5mm it doesn't work. It does however if you do two 4.99mm radii.

Would be nice if something like that worked :)