realthunder/FreeCAD_assembly3 #1018: [Question] New object doesn't not goes in to actived part container

ngocminhdao88, 2022-01-11:
I did those following steps:
1. Create a part container and mark it active
2. Create some primitive objects in part workbench

What i expected, those primitive objects will be automatically move to the part container. It does not do that. Is this a intended behavior? Is there an option or hack i can make, that new created objects will be moved to an active container (part in this case)?

[image]

I am using the latest release:
> OS: Manjaro Linux (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

bambukouk, 2022-01-11:
`Std Part is not used for modelling, just to arrange different objects in space`

[https://wiki.freecadweb.org/Std_Part](url)

just click on your sphere and cone and drag them with your mouse into the standard part container

ngocminhdao88, 2022-01-12:
Thank you, bambukouk.

It is indeed confusing.

bambukouk, 2022-01-12:
You are welcome!
You can now `close` this issue :+1:

closed 2022-01-12 by ngocminhdao88