realthunder/FreeCAD_assembly3 #1020: draft move not working.

sedatkrn, 2022-01-11:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Turkish/Turkey (tr_TR)

Open New file
i selected daraft or arch wb
i did work plane top
crate a poligon
selleced it
click to move command but not run.

[image]

sedatkrn, 2022-01-11:
00:02:12  flag
00:02:12  flag2
00:02:15  2.56786 <asm3.main> init_gui.py(14): no solver backend found
00:02:45  Pick a face, 3 vertices or a WP Proxy to define the drawing plane
00:02:55  ----------------
00:02:55  GuiCommand: Rectangle
00:02:55  Pick first point
00:02:55  Pick opposite point
00:03:01  ----------------
00:03:01  GuiCommand: Move
00:03:01  [(<Part::Part2DObject>, 'Face1')]
00:03:31  ----------------
00:03:31  GuiCommand: Move
00:03:31  [(<Part::Part2DObject>, 'Face1')]
00:03:36  ----------------
00:03:36  GuiCommand: Move
00:03:36  [(<Part::Part2DObject>, 'Face1')]
00:04:09  GuiCommand: Move
00:04:09  ----------------
00:04:09  GuiCommand: Move
00:04:09  [(<Part::Part2DObject>, 'Face1')]
00:04:09  Running the Python command 'Draft_Move' failed:
Traceback (most recent call last):
  File "C:\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\Mod\Draft\draftguitools\gui_move.py", line 75, in Activated
    self.get_object_selection()
  File "C:\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\Mod\Draft\draftguitools\gui_move.py", line 80, in get_object_selection
    return self.proceed()
  File "C:\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\Mod\Draft\draftguitools\gui_move.py", line 101, in proceed
    self.ui.lineUi(title=translate("draft", self.featureName), icon="Draft_Move")

lineUi() got an unexpected keyword argument 'icon'GuiCommand: Move
00:05:31  ----------------
00:05:31  GuiCommand: Move
00:05:31  [(<Part::Part2DObject>, 'Face1')]
00:05:31  Running the Python command 'Draft_Move' failed:
Traceback (most recent call last):
  File "C:\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\Mod\Draft\draftguitools\gui_move.py", line 75, in Activated
    self.get_object_selection()
  File "C:\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\Mod\Draft\draftguitools\gui_move.py", line 80, in get_object_selection
    return self.proceed()
  File "C:\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\Mod\Draft\draftguitools\gui_move.py", line 101, in proceed
    self.ui.lineUi(title=translate("draft", self.featureName), icon="Draft_Move")

lineUi() got an unexpected keyword argument 'icon'GuiCommand: Move
00:05:53  ----------------
00:05:53  GuiCommand: Move
00:05:53  [(<Part::Part2DObject>, 'Face1')]
00:05:53  Running the Python command 'Draft_Move' failed:
Traceback (most recent call last):
  File "C:\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\Mod\Draft\draftguitools\gui_move.py", line 75, in Activated
    self.get_object_selection()
  File "C:\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\Mod\Draft\draftguitools\gui_move.py", line 80, in get_object_selection
    return self.proceed()
  File "C:\FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111\Mod\Draft\draftguitools\gui_move.py", line 101, in proceed
    self.ui.lineUi(title=translate("draft", self.featureName), icon="Draft_Move")

lineUi() got an unexpected keyword argument 'icon'Retrying to obtain clipboard.
00:06:10  Retrying to obtain clipboard.
00:06:25  Retrying to obtain clipboard.
00:06:27  Retrying to obtain clipboard.
00:06:33  Retrying to obtain clipboard.

realthunder, 2022-01-12:
Please try this command using `Daily` branch release image and see if it works fine. I'll fix it for `Stable` soon.

sedatkrn, 2022-01-12:
i used to Daily branch. draft move is working nice.

closed 2022-01-12 by luzpaz